wled/WLED #3689: Lock LED Preferences and Segments tab

blargh2947, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
I provide WLED devices to friends and family who aren't tech savvy.  On occasion they will make changes in LED Preferences or Segments which change the configuration of the device, sometimes making the lights unusable.

**Describe the solution you'd like**
Add a configuration tick box that disables  changes to LED Preferences and/or Segments

**Describe alternatives you've considered**
It is currently possible to download the preferences and settings, however that requires being on the same network as the device itself which isn't always possible to quickly do.

There's a "settings pin" on security & macros but it doesn't appear to change any behaviors on the device. 

**Additional context**
A change to LED Preferences, or segments can cause behavior in the lights which is not expected.  For example changing WS28xx to SK6812 will cause the effects and colors to work incorrectly.

Also it's possible to create segments and save presets, but not know how to revert them.

Disabling the ability to make changes to these will make it easier to support devices when I give them to people, but still let them use the palettes, effects and colors.

softhack007, 2024-01-15:
Hi, just a question to better understand what would be needed:

> disables changes to LED Preferences and/or Segments

WLED Segments are "volatile" - technically speaking they are created/deleted on the fly.
Segments are saved in Presets, that's the only way to keep them over a reboot.

a) To really protect segments, it would be necessary to write-protect some presets (PIN or password for changing), at least write-protect the boot preset.

b) Another possibility could be to have a backup of presets.json and cfg.json on the WLED device. We would simply add a button for restoring these two files from the "on-device" backup, so user can recover if they screw up their settings.

blazoncek, 2024-01-15:
Segments are always dynamically configured and can be saved into presets. We will not change this behaviour.
As for settings, PIN and passphrase can be used to lock settings and/or updates. Both work as expected.
PIN has a *cooldown* timeout of 15min.

blargh2947, 2024-01-15:
What I didn't realize in the pin behavior, is that the device needs to be rebooted in order to lock the settings.

and once unlocked it stays that way until the next device reboot.

@softhack007 the concern with the backups, is that if the user isn't advanced enough to know "don't touch these settings" I think it might be too much to ask to suggest they backup/restore settings.

Can close this based on feedback I guess.

closed 2024-01-15 by blargh2947

blazoncek, 2024-01-15:
> What I didn't realize in the pin behavior, is that the device needs to be rebooted in order to lock the settings.

It does not. You just need to wait 15 min without clicking in UI.