wled/WLED #3691: 0.14.1 - missing ESP32 / ETH / AudioReactive version

papperone, 2024-01-15:
### What happened?

as per title there's no BIN for those 3 option (in the beta they are all present), what should I use then?

### To Reproduce Bug

no download for ESP32 standard, olny S2/S3/C3 version

### Expected Behavior

like any other release or even 14.1b3 I was supposed to find those 3 versions

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

papperone, 2024-01-15:
sorry my mistake... :(

closed 2024-01-15 by softhack007

softhack007, 2024-01-15:
Just for the curious - the files just moved down in the list a bit, as the new MCU variants al want their own firmware files 

[image]

papperone, 2024-01-15:
yes, by deafult only the first 12 files are show, and now we have more than 12 "flavours" ;)