wled/WLED #3692: ESP8685 settings lost after power cut

kub3let, 2024-01-15:
### What happened?

With support for ESP32-C3 in 14.x I flashed it onto a ESP8685-H4 (ESP32-C3 as QFN package)

While everything works as soon as the power is cut all settings are lost.

I already tried to erase flash and ota update but still the same.

Doing a reboot from within WLED keeps the settings, also on OTA update.

So I'm wondering does ESP8685 require another build ? AFAIK it's just a ESP32-C3 with 4mb built in flash

### To Reproduce Bug

Use https://install.wled.me/ to install onto a ESP8685-WROOM-06-H4

### Expected Behavior

Settings should persist

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-01-15:
Hi, 
Support for -C3 is still experimental in WLED. It means that some -C3 boards do work, some do not.

If you lose settings, this could indicate a problem with flash speed - our normal firmware uses "qio" (quad i/o) mode, maybe your board cannot run reliably with that.

To find out, you need to build & upload the firmware yourself, using the VSCode+platformIO development environment.

Add this line to the [env:esp32c3dev](https://github.com/Aircoookie/WLED/blob/a4a8e2614ea2b8479bb33fc53ac8ca2912f9df2c/platformio.ini#L473-L487) buildenv:
```ini
board_build.flash_mode = dio
```
Then build `env:esp32c3dev` and upload the firmware from VSCode. It might work. For further analysis, use serial monitor and check for crash information.

<br/>

Another possibility could be power supply - esp32 chips usually need at least 3.2V (stable!) power supply. If Vcc drops below this threshold, flash writing may fail and you end up with a corrupted flash filesystem, or even with an unusable device.

closed 2024-01-16 by kub3let

kub3let, 2024-01-16:
Little update on this, actually the hardware was at fault.

I was moding this WX300P led controller, replacing the unsupported CBU tuya module with a ESP32-C3 (ESP8685-WROOM-06-H4)

With some more testing I noticed strange light effects which turned out to come from the microphone, the controller has a dedicated music chip which does these effects and seems to be directly wired to the mosfest, It probably can be disable by pulling some gpio high or low but I did not want to bother with this software wise.

I simply desoldered the music ic and suddenly wled kept it's settings as well.

[image]

[image]