wled/WLED #3693: Since 0.14.1: Turning a segment off no longer fades to off, but turns off instantly

fedus, 2024-01-16:
### What happened?

On 0.14.0, turning a segment off would let the LEDs fade out. Since updating to 0.14.1, turning a segment off instantly turns the LEDs off.

### To Reproduce Bug

On default settings:
 - turn a segment on: LEDs fade to ON
 - turn the segment off: LEDs go off instantly

### Expected Behavior

That the LEDs fade to off

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 build 2401141

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Tested with 2-3 active segments. Haven't tested behaviour when only a single segment is defined.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-16:
Found & fixed.
Thanks for reporting.

fedus, 2024-01-16:
Wow, that was fast. Many thanks for your amazing work and the wonderful project!

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek