wled/WLED #3694: WS2812D Question, not really a bug

Doyle4, 2024-01-16:
### What happened?

Colors incorrect

### To Reproduce Bug

Select white etc

### Expected Behavior

Colors to work correct (Not WLEDs problem)

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32-S2

### Anything else?

More of a question on compatible LED's.
I got some [these](https://www.aliexpress.com/item/4000720415203.html?spm=a2g0o.order_list.order_list_main.5.6b481802rFf9K5) I was wondering if anyone has used them and what LED settings they used please? If I select the WS281X and change color mode to RGB its the closest I can get to correct colors, but colors such as White/Orange/Yellow isn't working, the LED doesn't emit a mix of colors to create the color I have selected, they look as if they have turned off.

I have tried other LED profiles to no luck.
Had a google and I can't seem to find anything on them with WLED, any help would be great thanks.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Doyle4, 2024-01-16:
Ok, I just found this quote online without any other details if anyone understands it.

Someone had asked something similar on a reddit post 2years ago which is now not able to reply too, someone was creating a matrix from standard WS281X strips and some one replied with "Why not just use WS2812D-F5's?"

 Reply was "WS2812D requires an additional resistor and capacitor to function, whereas this board contains the capacitor on the back."

blazoncek, 2024-01-16:
Please use WLED [forum](https://wled.discourse.group) or Discord which have broader user base than Github.
Perhaps someone has successfully used them and will reply there.

dosipod, 2024-01-16:
@Doyle4 Alie specs are not the best and i would advise you to contact the seller prior to ordering . Still on the product page your linked and reviews they are listed as not needing resistor and capacitor 
[image]

I have also ordered something like those ( not labeled as ws2812d  on alie ) but they are addressable and worked without
resistor and capacitor  https://www.aliexpress.us/item/3256802451320866.html?gatewayAdapt=glo2usa

Might be better to close this and use the forum or discord

closed 2024-01-16 by Doyle4