wled/WLED #3697: Color correction

MMMXXXZZZ, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
A lot of LED fixtures have bad CRI ratings and can't reproduce orange properly.

**Describe the solution you'd like**
A section in LED PREFERENCES that has color correction through use of Curves.
[image]

**Describe alternatives you've considered**
A more intuitive but may be harder to code solution is HSL Tuning as it allows the user to go through the list of Hue's and easily spot the faulty color and quickly start tuning in the correct color.
[image]

**Additional context**
In my specific case, I have a great silicone-diffuser strip that has lots of difficulty reproducing orange. In order to produce it, the strip must be driven in an extremely tight hue range, something between 1° and 3°, anything past that produces a green-ish yellow and 0° is pure red. This means that preset rainbow gradients in any controller jump from red to green-ish yellow in a very short distance. Strips with bad color calibration can be tuned up to at least 80% by the controller. As you can imagine, the white balance option does not work well enough and a permanent solution is required, after which white balance correction can be applied.

MMMXXXZZZ, 2024-03-27:
Sorry for digging this up, hoping to get some traction :)

agmorpheus, 2024-04-22:
Something similar to this would be awesome for correcting the white for RGB strips. My white on my WS2812b looks pink. I would be awesome to have an option tu reduce the blue part when the color is set to white.

TobyTredger, 2025-02-19:
Commenting to show interest.

blazoncek, 2025-11-14:
Using LUT's would be easy to implement in the backend (UI is different story). However they would need to be implemented in a similar way as Color Order Overrides to be truly universal. And even then it would be user's responsibility to apply correct LUT to particular set of pixels.
IMO not feasible with the current state of "lack of resources" on ESP (other than S3).

DedeHai, 2025-11-14:
actually: how about just adding an overlapping segment, make it a static color and "subtract" or "multiply" ? not the nicest or most convenient solution, but a viable workaround for the time being.

pdaoust, 2026-11-12:
For those looking for a crude solution that works decently:

1. Create a segment that overlaps your entire strip.
2. Set its blend mode to overlay, screen, or multiply.
3. Set it to a solid colour, play around with it until things look the way you like them.

Note that this doesn't answer what the OP is asking for -- it doesn't let you play around with arbitrary points on the curve, and depending on the blending mode and the effects applied to lower layers, it doesn't look quite right at colour extremes. But it may be helpful for some people. I'm using it in place of the white balance slider, which skews too much toward yellow and not enough toward white on my WS2812B strip (or maybe it's SK6812; you never can tell)

Note on blending mode choices:

* Multiply with an off-white correction colour should in theory be the right choice, but for extreme correction colour choices it seems to make it too dark.
* Screen is usually the lightest touch with the best effect, although you might have better mileage with overlay.

I've just started experimenting with this today, so i don't have a lot of deeply helpful advice. Just thought it might be helpful for people who are waiting for something more advanced and built-in.

DedeHai, 2026-11-12:
I have been working on color correction using XYZ transformation. It is nowhere near finished but I do have working test code as a rough sketch. So stay tuned ;)