wled/WLED #3700: Hyperion not handing brightness control back to WLED

bcootner, 2024-01-18:
### What happened?

I have been using WLED on a DigQuad to drive an led strip and sync to a raspi running Hyperion version 2.0.15 (all of these are controlled via Home Assistant but for the test I used WLED and Hyperion directly).

After updating to WLED to 0.14.1 when I turn on "Capture USB-Input" from Hyperion it works as expected however when I turn off "Capture USB-Input" I can no longer fully control the light from WLED. I can change the color and effects but the brightness and the on/off functionality don't work until I reboot WLED.

### To Reproduce Bug

 - Have an instance of WLED using 0.14.1 added to Hyperion
- Toggle "Capture USB-Input" on
 - Toggle "Capture USB-Input" off 

### Expected Behavior

After "Capture USB-Input" is off you can control the lights through WLED like normal

### Install Method

Self-Compiled

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ihavenonick, 2024-01-18:
I control an Ambilight via Hyperion with an ESP32 under WLED 0.14.1 and cannot confirm this.
If Hyperion or USB Capture is turned off, I have full control over WLED.

blazoncek, 2024-01-18:
Please see comments in #3054

jimmeh1988, 2024-01-19:
Something has definitely changed in 0.14.1. When turning off my TV, the lights would not turn back on as they have done before. Only when opening the app on my iPhone, does WLED seem to realise it's no longer receiving live information from Hyperion and then the lights come back on. Rolling back to 0.14.0, the issue goes away.

ihavenonick, 2024-01-19:
delete the ESP flash completely. After the update I have 100% the same behavior as with the previous version.

jimmeh1988, 2024-01-20:
I tried that and still seeing the same issue. I respect that you are seeing the same behaviour @ihavenonick but I am not...

ihavenonick, 2024-01-20:
I have two Hyperion systems running. Under Windows 10 and as Ambilight on the satellite receiver under Enigna 2, with both I can control everything manually if I close Hyperion or overwrite Hyperion in the WLED WebIF.

YaBoiDan, 2024-01-23:
I am also having similar issues. I can fully control WLED, but once Hyperion takes control, I can no longer turn off the LEDs, even when live override is enabled. Even after Hyperion is switched off, control is not restored until the device is rebooted. 

A factory reset does not correct this. Rolling back to version 14 from 14.1, the issue goes away. 

My setup is Hyperion running on an LG TV, PicCap as an input and WLED running on a QuinLED-Dig-Uno. Happy to test.

TheHolyRoger, 2024-01-29:
Might be the same issue as https://github.com/Aircoookie/WLED/issues/3680

Try my fix: https://github.com/Aircoookie/WLED/pull/3731

blazoncek, 2024-02-12:
Please use 0.14.2-b1

YaBoiDan, 2024-02-20:
Unfortunately, this did not remedy the issue for me, same issue, lights still on following turning off via light toggle still.

adix992, 2024-03-06:
Fix inc,

In Home Assistant disable all the light entities in the Hyperion Intergration. Use WLED integration for the light entity.
You can still use the switches in the Hyperion Intergration to turn on and off capture.

It's the light entity from the Hyperion Intergration that is overriding the brightness and on/off state for WLED after initial connection has been made

Hope this 

Hope this helps :)helps :)

adix992, 2024-03-07:
Nevermind,

For some reason this is not working anymore, not sure what I did differently yesterday.

A temp fix I have found is that selecting a WLED preset restores control to WLED without having to restart

quixotic120, 2024-03-16:
also having similar issues with same setup as @YaBoiDan except running on esp32 on 0.14.1. if hyperion does not "clean exit" and leaves WLED in a state where the lights are still thinking that they are receiving UDP data and I manually override I can manipulate the lights, change colors, brightness, etc but sending more UDP stream to them just turns them all off. Additionally unable to power off when in this state, clicking the button does nothing. going into the webui and rebooting fixes and acts normally. Rolling back to 0.14 resolves issues. 0.14.2-b1 also did not help

blazoncek, 2024-03-16:
Send `{"live":false}` via JSON API (HTTP POST).

closed 2024-03-22 by blazoncek