wled/WLED #3701: SK6812 not working after 0.14.1 update

KaSroka, 2024-01-18:
### What happened?

After updating to 0.14.1 my SK6812 RGBW 12V strip is not turning on anymore.

### To Reproduce Bug

Using SK6812 RGBW 12V strip update from 0.14.0 to 0.14.1. The strip will not work anymore. Downgrade back to 0.14.0 and the strip works as expected.

### Expected Behavior

After updating to 0.14.1 SK6812 RGBW 12V strip should work exactly the same as in 0.14.0

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-18:
Even though it shouldn't matter, did you erase flash and try to set up from scratch?

KaSroka, 2024-01-18:
If you mean full re-flash then no, I haven't tried this as my controller is hidden quite well.

blazoncek, 2024-01-18:
SK6812 is working just well on all my installs and dev set-ups.

ocean75, 2024-01-18:
Just to add to the thread. I had the same problem as discussed above when upgrading to 14.1. The lights did not turn on after upgrading. I had had to revert back to 14.0 and everything worked again.

blazoncek, 2024-01-18:
Reverting does not help solve the issue. 😄

ihavenonick, 2024-01-18:
I have the same configuration of ESP32 S2mini and SK6812 12V running and no problems after the update to 0.14.1

bokub, 2024-01-19:
I had the same problem, SK6812 led strip on a ESP8266 device.

The problem was solved after rebooting 3 times :shrug:

krisnoble, 2024-01-19:
I'm having issues with SK6812 on ESP8266 under 0.14.1 as well - loss of power and brightness controls although HA seems to think everything is fine so I'm guessing a fault in how WLED actually controls the strip.

By any chance are you using them in live mode? I was trying to diagnose thinking the live mode might be the issue (I have two running Hyperion and one which boots into a live mode preset) but not sure as coincidentally my devices running live mode are also my devices running the same strip/controller setup. 

My SK6812s running off an ESP32 are okay though, and my WS2812s on 8266s are all fine so looks like it's this specific combination that is having issues, possibly in conjunction with live mode?

softhack007, 2024-01-19:
> Is there any way to get debug logs to we can help you figure out what's wrong?

There is a short description for it to the release page.
* https://github.com/Aircoookie/WLED/releases/tag/v0.14.1
* scroll down to the "debug builds" section

Wombosvideo, 2024-01-20:
I have a physical on/off button which correctly updates the state in Home Assistant but it does not turn off the LEDs. The default preset works fine however, I just can't turn of the strip anymore unless disconnecting power

Wombosvideo, 2024-01-20:
Installed the debug build and waited for the debug info and then tried to turn off and turn back on the LEDs, without success (LEDs kept having 100% brightness).

Debug Log:

```
18:59:35 ---DEBUG INFO---
18:59:35 Runtime: 180078
18:59:35 Unix time: 1705773575,077
18:59:35 Free heap: 191616
18:59:35 Wifi state: 3
18:59:35 State time: 0
18:59:35 NTP last sync: 999000000
18:59:35 Client IP: 172.19.2.16
18:59:35 Loops/sec: 1657
18:59:35 Loop time[ms]: 3/40
18:59:35 UM time[ms]: 0/1
18:59:35 Strip time[ms]: 0/21
18:59:35 Segments: 1 -> 68B
18:59:35 Modes: 4*187=748B
18:59:35 Data: 4*187=748B
18:59:35 Map: 2*0=0B
18:59:43 Not-Found HTTP call:
18:59:43 URI: /presets.json
18:59:43 WS FileRead: /presets.json
18:59:43 Loop took 39ms.
18:59:43 Usermods took 0ms.
18:59:43 Strip took 0ms.
18:59:44 JSON buffer locked. (17)
18:59:44 JSON buffer size: 2420 for request: 3
18:59:44 JSON content length: 1407
18:59:44 JSON buffer released. (17)
18:59:48 JSON buffer locked. (11)
18:59:48 JSON buffer released. (11)
18:59:48 JSON buffer locked. (12)
18:59:48 JSON buffer size: 2407 for WS request (1406).
18:59:48 heap 191616
18:59:48 Sending WS data to multiple clients.
18:59:48 JSON buffer released. (12)
18:59:49 JSON buffer locked. (11)
18:59:49 JSON buffer released. (11)
18:59:49 JSON buffer locked. (12)
18:59:49 JSON buffer size: 2407 for WS request (1407).
18:59:49 heap 191620
18:59:49 Sending WS data to multiple clients.
18:59:49 JSON buffer released. (12)
```

tomlut, 2024-01-21:
It's not just SK6812, my WS812B LED strip wont turn on using 0.14.1 either. Works fine on 0.14.0.

Wombosvideo, 2024-01-21:
No Idea if this is related but after updating my ESP8266 as well (through HA), It wouldn't even turn on anymore. When plugging it in, the power LED shortly lights up and turns back off, and WLED doesn't start. I tried connecting to serial over USB but it won't show up.

aidbish, 2024-01-22:
Having the same issue with WS2812b, reverted back to 14.0 fixes it. using the push button to turn on/off works, unable to turn on/off using HA

casesolved-co-uk, 2024-01-23:
> It's not just SK6812, my WS812B LED strip wont turn on using 0.14.1 either. Works fine on 0.14.0.

I'm having the same issue with ESP8266 and WS2812 LEDs only on Bus 0.

To expand the symptoms:
* If I put the WS2812's on a Bus that is not zero, it works.
* If I put APA102's on Bus zero, they work

So the workaround seems to be:
* add a dummy LED on Bus zero: length 1 with a pin you are not using
* setup your LEDs on Bus one

For non-developers: Bus 0 == LED Output 1

Let us know if it works!! :)

Wombosvideo, 2024-01-23:
> So the workaround seems to be:
> 
> * add a dummy LED on Bus zero: length 1 with a pin you are not using
> * setup your LEDs on Bus one

This did not work for me

casesolved-co-uk, 2024-01-23:
> This did not work for me

Can you tell us exactly what you tried please? It will help debug the issue. Maybe screengrab the LED Output setup

ihavenonick, 2024-01-24:
Now I have problems too. I only switch via preset and after a certain time it happens that when I switch on, the ESP crashes and switches to default. When this state is reached, nothing can be reliably controlled using presets, as the ESP is very likely to crash. If I delete the ESP completely and reload the configuration, it works normally again for a while.
The workaround works, but you have to create all the presets again because saving the changes is also not possible. So it works, but the bug still exists.

casesolved-co-uk, 2024-01-24:
@ihavenonick can you post a screengrab of the LED output configuration please?

hughsaunders, 2024-01-24:
I have a wled instance with 339 SK6812s which became very unstable after the upgrade to 0.14.1. Downgraded back to 0.14.0 and ok so far. 

Similar to @ihavenonick, I only use presets, they are triggered by Home Assistant automations. 

Added a screenshot of my LED Seetings:
[image]

ihavenonick, 2024-01-24:
You don't use the workaround, so you have exactly the same problem with 0.14.1
Downgrade to 0.14.0 or use the workaround (downgrade is easier)
[image]
[image]

Wombosvideo, 2024-01-26:
Having added a separate segment for the dummy did the job. Brightness control and on/off is now back to "normal".

casesolved-co-uk, 2024-01-31:
> Even though it shouldn't matter, did you erase flash and try to set up from scratch?

This could be a config issue. It appears to be solved if I erase all before flashing.
If you can, try backing up your configs, erase and flash v14.1

casesolved-co-uk, 2024-02-01:
> This could be a config issue. It appears to be solved if I erase all before flashing. If you can, try backing up your configs, erase and flash v14.1

I was wrong. Full erase before program does not fix this.

casesolved-co-uk, 2024-02-01:
@blazoncek This bug is to do with having multiple strips, but of different types:

- Bus0 WS2812's only - works
- Bus0 & Bus1 WS2812's - works
- Bus0 WS2812's & Bus1 APA102's - Bus 0 stops working (given the above info, Bus 0 can be either WS2812 or SK6812 and Bus 1 probably just needs to be a different type)
- Bus0 APA102's & Bus1 WS2812's - works

Please could you try to reproduce again?
I'm guessing this could be somewhere in PolyBus?

tomlut, 2024-02-01:
I only have one WS2812 strip on Bus0, does not work on 14.1, does work on 14.0.

tomlut, 2024-02-01:
Yes I have a preset. Can't test right now heading out the door. Will get back to you later.

blazoncek, 2024-02-02:
>     4. Bus0 APA102's & Bus1 WS2812's - works

Unfortunately, that is the only combination that can work due to the way WLED organises buses.
If you are so kind, please update the documentation to clarify it.

LEDs with Data/Clock wires (APA102 & co.) can only be attached to bus 0.

KaSroka, 2024-02-02:
I only have SK6812 on bus 0 with no presets and it works on 14.0 but not on 14.1

casesolved-co-uk, 2024-02-02:
> > ```
> > 4. Bus0 APA102's & Bus1 WS2812's - works
> > ```
> 
> Unfortunately, that is the only combination that can work due to the way WLED organises buses. If you are so kind, please update the documentation to clarify it.
> 
> LEDs with Data/Clock wires (APA102 & co.) can only be attached to bus 0.

Then I've no idea what the cause of this bug is, but there are too many people with it to simply mark it as `cannot reproduce`.

I've tried to narrow it down but I only have one lot of WS2812's and no SK6812's, all my other LEDs are APA102 (data&clock) so it's impossible for me to get to the bottom of, especially with this new undocumented limitation.

The things I would try on v14.1 are:

- Delete and recreate presets
- Delete, save and then reconfigure your LED outputs (i.e. force a re-save)
- Erase and reflash the chip from scratch (probably won't help)

I have a feeling it has something to do with the json corruption bug that was fixed by @willmmiles meaning that the json stored on v14.0 is corrupt but somehow works. But then is read as corrupt by v14.1. Just a guess.

blazoncek, 2024-02-02:
> I only have SK6812 on bus 0 with no presets and it works on 14.0 but not on 14.1

Factory reset on 0.14.1 and recreate presets from scratch and report back.

tomlut, 2024-03-20:
Just had to re-install as my ESP32 died after a power outage. 

No pre-sets configured. 

WELD fails to work correctly with Hyperion using 14.2,

It works after the first install then after I turn in Hyperion off the WELD strip can not be turned on again. 

Using 14.0 everything works. On off on off. No problem.

bokub, 2024-03-25:
If I can bring additional information, I still have the same issue after updating to 14.2.0

tomlut, 2024-03-25:
Tested with a new ESP board. This occurs for a new install.

blazoncek, 2024-03-25:
FYI I have SK6812 (and plenty of other chip variants from APA102 to UCS9904 and WS281x all variants) on various ESPs, ESP01, ESP8266 Lolin D1 mini, ESP32, ESP32-C3 and ESP32-S2 as I need to test them before releasing any version and they do work as expected on each and every device.
IMO you should check your configuration, wiring and other electrical characteristics, including PSU and level shifters.

tomlut, 2024-03-25:
There is nothing wrong with the wiring as evidenced by the fact that **it works perfectly fine on 14.0**

blazoncek, 2024-03-25:
Then do not update. 🤷‍♂️

tomlut, 2024-03-25:
That is exactly what I have done. However it would be beneficial to get this resolved.

blazoncek, 2024-03-25:
There is nothing to be resolved as far as I can see.
I am sorry but none of my tests were unsuccessful and I cannot see any change in code that would suggest otherwise.

tomlut, 2024-03-25:
Did you do your tests with the sync interface enabled?

Did you test turning off via the sync interface?

blazoncek, 2024-03-25:
I do not recall every test I made but you can be sure I do plenty.
And BTW do not get off-topic. OP is "SK6812 not working" not sync issues. Open a new issue if sync isn't working for you.

tomlut, 2024-03-25:
Done. https://github.com/Aircoookie/WLED/issues/3854

KaSroka, 2024-04-21:
I can confirm it's fixed on 14.3 for me.

softhack007, 2024-04-21:
Thanks @KaSroka for the confirmation. As Sk6812 is working (same behaviour as ws2812) I think we can close this issue now. I can also add that my sk6812 LEDs are working. 

Please note that "white mangement" was fixed after 0.14 release, so depending on "white mode" the white channel behaves slightly different (but more correct) now.

closed 2024-04-21 by softhack007