wled/WLED #3706: S2 mini support Use alternative Bootloader (S2 only)

incarvr6, 2024-01-19:
### What happened?

Still not possible to install

### To Reproduce Bug

Use alternative Bootloader (S2 only)

### Expected Behavior

install

### Install Method

Binary from WLED.me

### What version of WLED?

sk6812

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

Now that on https://install.wled.me/  for 14.1 there is new option Use alternative Bootloader (S2 only)
Still not possible to install

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

incarvr6, 2024-01-19:
Now that on https://install.wled.me/ for 14.1 there is new option Use alternative Bootloader (S2 only)
Still not possible to install

blazoncek, 2024-01-19:
Apparently there are several versions of S2 hardware and we cannot support all of them (as we do not have each and every development board there is).
Please research the capabilities on your own (for your hardware) and submit feedback and possible resolution.

incarvr6, 2024-01-19:
This is the board in question: https://www.wemos.cc/en/latest/s2/s2_mini.html Market is flooded with them.
At least there is some progress for ESPhome, it is possible to flash them with esptool. https://community.home-assistant.io/t/esp32-s2-and-esphome-upload/382885/68

blazoncek, 2024-01-19:
> This is the board in question: https://www.wemos.cc/en/latest/s2/s2_mini.html Market is flooded with them.

I have that particular board myself and it is working flawlessly once bootloader, partition map and WLED firmware are uploaded to it. Please research your flashing procedure and use esptool.

incarvr6, 2024-01-19:
Using Tasmota https://tasmota.github.io/install/ use device Tasmota32-CDC (dfu mode) i was able to flash it, then after connecting to network running OTA update with wled s2 bin file, now it is finally flashed WLED!!!! I haven't try to connect LED's yet but at least finally it is running WLED.

blazoncek, 2024-01-19:
Please close then.

incarvr6, 2024-01-19:
Well even if i found solution for workaround flashing from https://install.wled.me/ still doesn't work, even after flashing with Tasmota.
Same is with esptoolI, suppose problem is with bootloader, partition map.

blazoncek, 2024-01-19:
As noted above experiment with flash options and flashing procedure. Not WLED firmware issue.
https://docs.espressif.com/projects/esptool/en/latest/esp32s2/esptool/flash-modes.html

incarvr6, 2024-01-19:
Bug is still on https://install.wled.me/ it is not recognising initializing S2 board! Firmware works, but only if installed on Tasmota flashed device. If they could make web flasher work then wled should be able to do as well.

blazoncek, 2024-01-19:
Open an issue on installer repo, not here.

esptool is official Espressif tool for flashing. You need to provide correct bootloader and correct partition map for your board.
PlatformIO does that for you if you compile yourself.

As a matter of fact, the alternate bootloader on https://install.wled.me was provided by PlatformIO and was tested by several users (including me) to work correctly.

incarvr6, 2024-01-19:
Since S2 firmware it self is operational i'm closing this

closed 2024-01-19 by incarvr6

blazoncek, 2024-01-19:
You may want to try lowering flash speed from 80MHz to 40MHz.

mike-goldfinger, 2024-02-14:
I was able to successfully flash WLED_0.14.1_ESP32-S2.bin on my ESP32-S2 (ESP32-S2FNR2 (revision v1.0)) device by following these steps:

1. Installed release Tasmota32-cdc via https://tasmota.github.io/install/
2. Connected to the TasmotaXXXX Access Point.
3. Configured WLAN to my Access Point.
4. Downloaded WLED_0.14.1_ESP32-S2.bin and flashed it via the Tasmota web UI using "Upgrade by file upload".
5. Connected to the WLED-AP.
6. Configured WLAN to my Access Point again.

spacecdr, 2024-03-19:
Mine exactly by:
esptool.py --port /dev/cu.usbmodem01 --before no_reset --after no_reset --baud 115200 --chip esp32s2 write_flash 0x0 tasmota32s2cdc.factory.bin

skd2314, 2025-03-16:
My method for S2-mini:
1. https://adafruit.github.io/Adafruit_WebSerial_ESPTool
2. esptool.py write_flash 0x01000 [S2_bootloader.bin](https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/S2_bootloader.bin)
3. esptool.py write_flash 0x08000 [S2_partitions_4M.bin](https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/S2_partitions_4M.bin)
4. Config wifi [WLED](https://install.wled.me/)

[image]

[image]