wled/WLED #3708: Node list is not real links, I see no reason for these to be js

tablatronix, 2024-01-20:
I see no reason why these links are coded like this.
I cannot open in new target, copy link or anything logical for an external link.

[image]

WoodyLetsCode, 2024-01-22:
You're right! I'll fix this.

blazoncek, 2024-01-22:
Do not forget that some mobile apps (which utilise web view) may not support everything regular browser can.

tablatronix, 2024-01-22:
Yeah thats why I said styleguide, not sure if there was a reasoning but not to break the wled native app, surely js href is a thing tho?

blazoncek, 2024-01-22:
> Yeah thats why I said styleguide, not sure if there was a reasoning but not to break the wled native app, surely js href is a thing tho?

Nobody is prohibiting you from submitting a PR. 😉 There are just too many users and cases and too few developers.

tablatronix, 2024-01-22:
I dont know if this was done on purpose or what the coding standards are for this, its cleary not just a simple webapp. 

https://github.com/Aircoookie/WLED/blob/main/CONTRIBUTING.md is just just coding style not UI standards.

Is wled native just wrapping the webapp or using js api and its own ux?

WoodyLetsCode, 2024-01-22:
WLED Native is basically just a browser (WebView) that opens the web UI of WLED.
WLED Native also exists for iOS.
I don't know what UI standards you would expect, but you can basically just modify the code and then make a PR.
If you are unsure about how to implement something, you could also ask on the WLED Discord server. 😄 
I will later fix this issue here, so you know, and we don't do double work.
If you want to contribute to WLED anyway, you are very welcome to do so. 🤗

tablatronix, 2024-01-22:
I notice NO ui elements are aria safe, so I guess this is not really a small problem , SO I am going to change this request to just make it eeasier to OPEN NODES in a new window or copy ip address only.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek

tablatronix, 2024-03-22:
Will check it out tonight!

PR #3715 (2024-02-01): Allow nodes buttons to be opened in new target or copy ip address
This makes nodes buttons to proper href links.
This fixes #3708.