wled/WLED #3710: Cannot select RGB pin on Atom S3 Lite

rulkens, 2024-01-21:
### What happened?

I successfully installed WLED on the Atom S3 Lite. When configuring the LED settings I wanted to use the inbuilt LED as the only LED to be controlled, for debugging purposes. The LED is connected to the GPIO35 pin. When setting the pin on the settings page, it would revert to the previously set value.

### To Reproduce Bug

Go to the settings page. Select pin 35 as the input pin for LED input channel 1. Click the save button, and visit the led settings page again. The value will not be saved.

### Expected Behavior

I would like to be able to select and save pin GPIO35 as an led pin on the ESP32S3. This should be possible.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Anything else?

So, I went through the C++ code, and found the `PinManagerClass::isPinOk` function. There, it seems to do a check for the ESP32 and on line 196 it checks if the pin is defined as an input and the pin number is 40 or higher. Only then does it return true. In case of the ESP32S3, I assume (some of) these pins should also be allowed to be defined as output pins:

```
if (gpio < 40 && !output) return true; //34-39 input only
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-01-21:
Hi

> found the `PinManagerClass::isPinOk` function. There, it seems to do a check for the ESP32 and on line 196

That line is for classic esp32. For -S3 this part is used:
https://github.com/Aircoookie/WLED/blob/74fed5695f2d76caa84bf4f6258fc5141d822cb1/wled00/pin_manager.cpp#L247-L252

If you use a firmware with PSRAM support, then actually WLED reserves GPIO for PSRAM and you cannot use them.

https://github.com/Aircoookie/WLED/blob/74fed5695f2d76caa84bf4f6258fc5141d822cb1/wled00/wled.cpp#L352-L355

If you don't have a PSRAM build, then most likely the pin you want is already configured somewhere else in WLED. Check all settings pages, including usermod settings.

rulkens, 2024-01-21:
Thanks @softhack007! I got the AtomS3 Lite working now with a custom `platform_override.ini` with the following details:

```
[platformio]
default_envs = atoms3lite

[env:atoms3lite]
extends = env:esp32s3dev_8MB
build_flags = ${env:esp32s3dev_8MB.build_flags} ${common.debug_flags}
```

Flashing with PlatformIO works without problems.

closed 2024-01-21 by rulkens