wled/WLED #3712: Some effects are not working with "sparse" LED matrix

imeszaros, 2024-01-22:
### What happened?

In my clock project I'd like to use the 2D features of WLED, so I set up a `ledmap.json` where my "display" is 17x7 matrix, but lot's of LEDs are missing so I put `-1` to the appropriate places (having 58 LEDs instead of 119 only).

The above works totally fine except that some effects refuse to work properly, since they update their output using the colors from the previous effect iteration. Since physically missing LEDs also do not have memory allocations, the previous color cannot be used by the effect to work properly.

### To Reproduce Bug

Create a matrix with one row "missing" i.e. with `-1` cells in `ledmap.json` and try to use the Matrix effect for example. Rows under the "missing" row will not be affected by the effect.

### Expected Behavior

All effects work with sparse matrices.

### Install Method

Self-Compiled

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I could work the problem around by setting the LED count to the theoretic number of LEDs (17x7=119) and filling the `-1` cells with numbers >58 incrementally, this way WLED allocates the necessary memory for all the LEDs and effects work properly, but the problem is that automatic brightness limiter now thinks of course that there are 119 LEDs instead of 58 so it estimates that more current is flowing and therefore it limits the brightness to significantly less value.

To be honest I do not know what the proper solution would be. Maybe `-2` in `ledmap.json` could mean that the pixel is phisically not there (do not consider it with ABL nor try to drive it i.e. do not shift out the value to the strip) but allocate memory for it for calculations.

Another workaround would be to allow an overriding LED count for the ABL feature. Whis probably makes things even more confusing... :)

I would happy to help with this if a decision is made about which direction to go to.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-22:
Known issue for the effects that rely on previous pixel values. For sparse matrices missing pixels will always produce *black* value.
There is no other way than to re-engineer such effects.

imeszaros, 2024-01-22:
Thanks for the confirmation. Just for reference, in `0.14.1` the following effects are affected (calling `SEGMENT.getPixelColor(...)` or `SEGMENT.getPixelColorXY(...)`):

- `mode_fireworks`
- `mode_rain`
- `mode_juggle`
- `mode_colortwinkle`
- `ripple_base`
- `mode_2Dgameoflife`
- `mode_2Dmatrix` - **fixed in 0.15**
- `mode_matripix`
- `mode_pixelwave`
- `mode_DJLight`
- `mode_freqmatrix`
- `mode_freqwave`
- `mode_waterfall`
- `mode_2DFunkyPlank`
- `mode_2Dsoap`