wled/WLED #3714: MQTT device/api topic

tech-byte, 2024-01-22:
### What happened?

Sending {"on":true} to the light/api topic doesn't update the status of device topic.

### To Reproduce Bug

Sending the json {"on":true} to the api topic won't update the device topic.

### Expected Behavior

Sending the JSON {"on":true} to the api topic should update the device topic with "ON".

### Install Method

Binary from WLED.me

### What version of WLED?

Reproduced on 0.14.0 and 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-22:
Perhaps improperly formatted MQTT message payload?
MQTT message goes through same codebase as regular HTTP JSON.

dosipod, 2024-01-22:
@tech-byte  can you add more details , i seen someone on discord ( might be you ) who wanted to see on/off response in mqtt , that is at the moment is not there as listed by the kb and the code only   4 topics . https://github.com/Aircoookie/WLED/blob/main/wled00/mqtt.cpp.  Other wise the response is there with ws when using {"on":true} or {"on":"t"} from mqtt 

[image]

Might be this is more of a feature request to add on/off topic  to mqtt

tech-byte, 2024-01-24:
Thanks for the reply. This issue I am facing is when using the group topic linking several lights to Home Assistant. With the group setup the status topic doesn't show any other state other than offline. I think the issue here is that the device topic isn't reflecting the actual state of the light. As per the below image below if the api topic is used to turn the light on (along with lots of other settings) the device topic remains in the off state. This confuses Home Assistant (and possibly other platforms) as won't apply scene information to the light as it thinks it is off. The work around at the moment is to publish to 2 topics. 1 to turn the device topic of wed/study to "ON" then the settings of JSON through wed/study/api

API on (device topic remains off)
[image]

dosipod, 2024-01-24:
@tech-byte That is what i am saying , there is no ON /OFF in mqtt response  , what you are seeing after the = is just the mqtt client last command so not an mqtt response  from wled  , the offline stataus is not also related to the unit on/off but only related to if the device is seen on the mqtt broker meaning it is offline or online from network point of view   , i hope the below video will make it clear and as said this is a feature request in my opinion 
[image]

closed 2024-07-23 by github-actions[bot]