wled/WLED #3717: Wled widgets buttons on/off

danielmosss, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
No, this is not related to a problem. 

**Describe the solution you'd like**
A 1x1 widget button where you can only press an on/off button. This will turn the ledstrip on or off. When you drag a widget on the screen for the first time you can configure it and have to select a ledstrip, color, brightness and effect. 

**Describe alternatives you've considered**
I couldnt found a widget app that could use api and get this feature. 

Thank you for your ideas for making WLED better!

benjamw, 2024-02-03:
You might have better luck posting this request to https://github.com/Aircoookie/WLED-App

blazoncek, 2024-02-03:
Official @Aircoookie app has not been updated in a while.
Currently the best alternative is @Moustachauve WLED-Native.

This is request for the app, not WLED itself. Please open an issue in the app repositories.