wled/WLED #3719: Emulate Alexa devices per segment

Titanium177, 2024-01-25:
**Is your feature request related to a problem? Please describe.**
Yes, i'm using two different stripes with 1 controller. Every Alexa command sets brightness etc. for both ones. The current workaround is to use presets, which limits it to preconfigured scenes.

**Describe the solution you'd like**
Add a option to emulate alexa devices per segment, just like it already exits for presets.

**Describe alternatives you've considered**
Adding separate controllers would be a way, but is unnecessary.

BennoB666, 2026-03-13:
I hope this feature ist comming. We can save a lot of controller hardware.

netmindz, 2026-04-09:
@coderabbitai Is this possible within the current rather basic hack that presents WLED as Hue in order to be controlled from Alexa and also add this feature in a backwards compatible manner?