wled/WLED #3720: v0.14.1: Brightness and Power are ignored when Transition is set to 0

fcmower, 2024-01-25:
### What happened?

If Transition is set to 0 in a preset or in the UI, neither
Brightness nor Power can be changed through json or the UI.
X.X.X.X/json/state and the UI show all changes, but visually, the
pixels do not change brightness (up, down, off, or on).

With a bootup preset, if the preset has transition:0, the preset
turns on to very low brightness, not the brightness saved in the
preset.

For example:
{"transition":1,"bri":255} Pixels are bright.
{"transition":0,"bri":10} Pixels stay bright.
{"transition":1,"bri":10} Pixels dim.
{"transition":0,"on":false} Pixels stay on.

I think this started in 14.0, but I'm not sure.  I remember
after a software update, some of my presets in my playlists
started behaving differently.  Today when I looked at my
presets.json, I noticed those misbehaving presets all had
transition:0.  I changed them to transition:1 and now they
work normally again.

Installed version: WLED 0.14.1
(ESP32-D0WDQ6 build 2401141)

### To Reproduce Bug

In the user interface, set Transition to the value 1.
Setup a solid color.
In the user interface, set Transition to the value 0.
Adjust the Brightness slider up and down.
Notice the brightness does not change.
In the user interface, turn off Power.
Notice the power does not turn off.
In the user interface, set Transition to the value 1.
Notice Brightness and Power function normally.

### Expected Behavior

Brightness and Power should function normally at any Transition setting greater than or equal to 0.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fcmower, 2024-01-25:
I forgot to point out, I also had transition:0 in my playlists that I changed to :1.  The playlist issue
is a little more confusing.  I don't know if I had to change both the preset and the playlist to :1 or
what.  I do know if a playlist entry has transition:0, the brightness is not updated and it keeps the
brightness setting from the previously played preset ... it's a little confusing. :-/

blazoncek, 2024-01-25:
This is supposed to be fixed in 0.15

blazoncek, 2024-01-25:
As a workaround, use 0.1s transition time.

closed 2024-06-19 by blazoncek