wled/WLED #3725: Almost every time I open setting page of 0.14.x my ESP8266 crashes and reboot.

andrehen, 2024-01-25:
### What happened?

After updating my ESP8266 Devices to v0.14.0 or v0.14.1 they are experiencing almost every time a crash while oppening "LED Preferences" page. I get the error `Loading of configuration script failed. Incomplete page data!`. The devices then resets to default preset.
I only get this error on ESP8266 devices while using iPhone (via wled-native app or Safari). **In a desktop browser I don't get this error**.... 
Also on ESP32 devices works fine with safari, had no issues with my ESP32 devices.
Downgraded my 8266s to v0.13.3 and no issues...
I have 3 8266 devices on use, they all show the same crash.

### To Reproduce Bug

Using Safari or wled-native app on iPhone and a ESP8266 device:

1. Open config, then LED Preferences. (expect device to crash and reboot to default preset)

If crash is not observable, make some changes, switch preset, then try again.

The crash can occur on other pages too, but "LED Preferences" is the one the almost aways trigger the crash.

### Expected Behavior

To open LED Preference page without rebooting/crashing

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-25:
Please post your configuration and screenshot of the info dialog.

andrehen, 2024-01-25:
This is my info dialog:

[image]

Config:
```json
{
  "rev": [1, 0],
  "vid": 2401141,
  "id": { "mdns": "", "name": "WLED Bar 60", "inv": "Light" },
  "nw": {
    "ins": [
      {
        "ssid": "WIFI_2GA05A4C",
        "pskl": 15,
        "ip": [0, 0, 0, 0],
        "gw": [0, 0, 0, 0],
        "sn": [255, 255, 255, 0]
      }
    ]
  },
  "ap": {
    "ssid": "WLED-AP",
    "pskl": 8,
    "chan": 1,
    "hide": 0,
    "behav": 0,
    "ip": [4, 3, 2, 1]
  },
  "wifi": { "sleep": true, "phy": false },
  "hw": {
    "led": {
      "total": 60,
      "maxpwr": 2000,
      "ledma": 55,
      "cct": false,
      "cr": false,
      "cb": 0,
      "fps": 42,
      "rgbwm": 255,
      "ld": false,
      "ins": [
        {
          "start": 0,
          "len": 60,
          "pin": [2],
          "order": 0,
          "rev": false,
          "skip": 0,
          "type": 22,
          "ref": false,
          "rgbwm": 0,
          "freq": 0
        }
      ]
    },
    "com": [],
    "btn": {
      "max": 2,
      "pull": true,
      "ins": [
        { "type": 2, "pin": [0], "macros": [0, 0, 0] },
        { "type": 0, "pin": [-1], "macros": [0, 0, 0] }
      ],
      "tt": 32,
      "mqtt": false
    },
    "ir": { "pin": -1, "type": 0, "sel": true },
    "relay": { "pin": -1, "rev": false },
    "baud": 1152,
    "if": { "i2c-pin": [-1, -1], "spi-pin": [-1, -1, -1] }
  },
  "light": {
    "scale-bri": 100,
    "pal-mode": 0,
    "aseg": false,
    "gc": { "bri": 1, "col": 2.8, "val": 2.8 },
    "tr": { "mode": true, "fx": true, "dur": 7, "pal": 0, "rpc": 5 },
    "nl": { "mode": 1, "dur": 60, "tbri": 0, "macro": 0 }
  },
  "def": { "ps": 8, "on": true, "bri": 128 },
  "if": {
    "sync": {
      "port0": 21324,
      "port1": 65506,
      "recv": {
        "bri": true,
        "col": true,
        "fx": true,
        "grp": 1,
        "seg": false,
        "sb": false
      },
      "send": {
        "dir": false,
        "btn": false,
        "va": false,
        "hue": true,
        "macro": false,
        "grp": 1,
        "ret": 0
      }
    },
    "nodes": { "list": true, "bcast": true },
    "live": {
      "en": true,
      "mso": false,
      "port": 5568,
      "mc": false,
      "dmx": {
        "uni": 1,
        "seqskip": false,
        "e131prio": 0,
        "addr": 1,
        "dss": 0,
        "mode": 4
      },
      "timeout": 25,
      "maxbri": false,
      "no-gc": true,
      "offset": 0
    },
    "va": { "alexa": false, "macros": [0, 0], "p": 0 },
    "mqtt": {
      "en": false,
      "broker": "",
      "port": 1883,
      "user": "",
      "pskl": 0,
      "cid": "WLED-c5d339",
      "rtn": false,
      "topics": { "device": "wled/c5d339", "group": "wled/all" }
    },
    "hue": {
      "en": false,
      "id": 1,
      "iv": 25,
      "recv": { "on": true, "bri": true, "col": true },
      "ip": [192, 168, 0, 0]
    },
    "ntp": {
      "en": false,
      "host": "0.wled.pool.ntp.org",
      "tz": 0,
      "offset": 0,
      "ampm": false,
      "ln": 0,
      "lt": 0
    }
  },
  "remote": { "remote_enabled": false, "linked_remote": "" },
  "ol": {
    "clock": 0,
    "cntdwn": false,
    "min": 0,
    "max": 29,
    "o12pix": 0,
    "o5m": false,
    "osec": false
  },
  "timers": {
    "cntdwn": { "goal": [20, 1, 1, 0, 0, 0], "macro": 0 },
    "ins": []
  },
  "ota": { "lock": false, "lock-wifi": false, "pskl": 7, "aota": true },
  "um": {}
}

```

blazoncek, 2024-01-26:
Nothing seems out of ordinary. Please monitor free heap size (less than 16kB means trouble) and/or get a debug build and monitor serial output for potential crash dumps.

Jerry80, 2024-01-26:
Hi,

I actually have similar problem. Bought this controller https://www.athom.tech/blank-1/wled-rgbcct-analog-and-digital-light-strip-controller with ESP8266 (FW 0.14.0-b1), everything worked great.
I updated to stable 0.14.1 version but controller stopped working (it was offline), so I had to unplug it couple times to get online but it didn't last too long. So I used 0.14.0 version, now it's ok but it's too slow, sometimes when I open wled-native app and click on controller, it's just black screen or it takes too long to open and actually everything I do takes too long. Originally I had 0.14.0-b1 version and it worked fine fast with no problems. So looks like beta version was much better than these stable versions, do you know why stable versions 0.14.0 and 0.14.1 have these kind of problems and how to fix it? Thanks

willmmiles, 2024-01-26:
I'll see if I can borrow an iPhone and reproduce this later.  Reliable crashes are so much easier to debug.

andrehen, 2024-01-26:
> Nothing seems out of ordinary. Please monitor free heap size (less than 16kB means trouble) and/or get a debug build and monitor serial output for potential crash dumps.

Installed the debug version, the heap size is always somewhere between 17,2 to 18 KB. I got this on console when the crash happened:

```
Sending WS data to multiple clients.
JSON buffer released. (12)
Loop delayed more than 3ms.
settings resp0
WS client disconnected.
---DEBUG INFO---
Runtime: 150087
Unix time: 1706300637,087
Free heap: 19392
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.0.127
Loops/sec: 606
Loop time[ms]: 7/64
UM time[ms]: 0/1
Strip time[ms]: 0/62
Segments: 1 -> 68B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40101cbd epc2=0x00000000 epc3=0x00000000 excvaddr=0x40035e69 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffeaf0 end: 3fffffb0 offset: 0150
3fffec40:  3fff817c 00000010 3fff8208 3fff6be4  
3fffec50:  00000000 3fff6c9c 00000020 40101f44  
3fffec60:  402cb1c8 00000010 00000000 4027ce95  
3fffec70:  00000000 00000218 00000020 4027cef4  
3fffec80:  3fff82cc 00000010 0000005a 4027cf24  
3fffec90:  3fff82cc 00000010 3fff82e4 40280678  
3fffeca0:  00000000 00003097 3fff888c 00000000  
3fffecb0:  4025a1a4 00000020 3fff817c 3fff6be4  
3fffecc0:  00000000 00000000 3fff6c9c 402809a2  
3fffecd0:  00000002 00000218 3fff8454 3fff6be4  
3fffece0:  00000000 3fff888c 00000218 00000218  
3fffecf0:  00000218 00000000 00000430 40101bc2  
3fffed00:  00000001 00000020 3fff82cc 00000000  
3fffed10:  3fff6c9c 3fff71d4 00000430 4024fbf4  
3fffed20:  3fff8454 3fff7b48 3fff7b48 40265751  
3fffed30:  00000000 3fffed00 3fffed60 00000430  
3fffed40:  3fff7b04 3fff71d4 0000011a 4024fc27  
3fffed50:  3fff7b04 3fff7b48 0000011a 4025b0fe  
3fffed60:  00000000 3fff7504 00000000 40265a7a  
3fffed70:  4025a1a4 3fff7504 3fff7b04 4025994c  
3fffed80:  3fff7504 00000000 00000316 3fff8454  
3fffed90:  69746365 00000000 3fff7504 00000002  
3fffeda0:  00000002 3fff7a51 3fff7504 4025b46d  
3fffedb0:  3fff7504 3fff7a51 401049c6 00000100  
3fffedc0:  3fff7a51 7fffffff 00002200 00000002  
3fffedd0:  00000002 3fff7a51 3fff7504 4025943a  
3fffede0:  3ffeb67c 00004288 000000ff 00000000  
3fffedf0:  00000001 00000020 3fff6c5c 40101f10  
3fffee00:  3fff7538 00000000 c49ba5e3 02a6a1c3  
3fffee10:  4028a974 3fff7724 3fff763c 3fff785c  
3fffee20:  3fffee70 3fff71d4 3fff6c9c 402502b0  
3fffee30:  000001b1 3fff78a2 3fff71d4 3fff7504  
3fffee40:  00000000 000005f9 3ffe899c 40101bc2  
3fffee50:  3fff000a 3fff30dc 3fff709c 00000000  
[... 371 lines cut ...]
NTP last sync: 999000000
Client IP: 192.168.0.127
Loops/sec: 756
Loop time[ms]: 2/25
UM time[ms]: 0/1
Strip time[ms]: 0/24
Segments: 1 -> 68B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Loop delayed more than 3ms.
Loop delayed more than 3ms.
---DEBUG INFO---
Runtime: 90041
Unix time: 90,041
Free heap: 19264
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: 192.168.0.127
Loops/sec: 757
Loop time[ms]: 3/14
UM time[ms]: 0/1
Strip time[ms]: 0/11
```

andrehen, 2024-01-26:
> Nothing seems out of ordinary. Please monitor free heap size (less than 16kB means trouble) and/or get a debug build and monitor serial output for potential crash dumps.

Actually, there is a better crashdump, with readable stack trace as I compiled myself in VSCode PlatformIO:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40101cbd epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002c6e1 depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40101cbd in umm_malloc_core at umm_malloc.cpp:?

>>>stack>>>

ctx: sys
sp: 3fffeba0 end: 3fffffb0 offset: 0150
3fffecf0:  0000003f 00000001 3fffed94 00000049  
3fffed00:  3ffea077 00000010 00000020 40101f44  
3fffed10:  3ffea077 00000010 3fffed70 402657fe  
3fffed20:  00000000 3fff8a04 3fff6e44 3fffed94  
3fffed30:  00000047 00000008 3fff8a04 00000049  
3fffed40:  3ffea077 0000000c 3fffed70 40265893  
3fffed50:  00000000 0000000c 3fffed70 402658d5  
3fffed60:  00000000 3fff7fc8 3fff7f94 402576cf  
3fffed70:  00000000 000d0000 00000000 00000000  
3fffed80:  00656461 00000000 00000000 00000048  
3fffed90:  00000048 3fff8a5c 003f003f 80000000  
3fffeda0:  65636341 00007470 06000000 40265c2e  
3fffedb0:  00000000 3fff817a 3fff41f0 00000030  
3fffedc0:  3fff817a 3fff3f34 3fff6ad8 00000049  
3fffedd0:  00000149 3fff817a 3fff7f94 40259432  
3fffede0:  00000014 3fff6ad4 000000ff 00000000  
3fffedf0:  00000001 00000020 3fff6f94 40101f10  
3fffee00:  3fff7fc8 00000000 1ced9168 0208b019  
3fffee10:  3fff3f34 3fff7f94 3fff88ac 3fff80cc  
3fffee20:  3fffee70 3fff88ac 3fff6ad4 402502a8  
3fffee30:  000001b1 3fff8112 3fff88ac 3fff7f94  
3fffee40:  00000000 0000054d 3ffe899c 40101bc2  
3fffee50:  3fff000a 3fff30dc 3fff6ad4 00000000  
3fffee60:  3fff80cc 3fff6ad4 3fff88ac 40250308  
3fffee70:  3fff6fb8 3fff6fac 00000000 3fff3f34  
3fffee80:  3fff41a8 3fff3f34 3fff80cc 3fff41f0  
3fffee90:  3fff6ad4 3fff41f4 3fff41f5 4028041d  
3fffeea0:  3ffecfd0 00000141 3ffecfed 3ffecfc4  
3fffeeb0:  00000018 00000000 3ffecfe0 3ffecffe  
3fffeec0:  00000000 3ffed0db 3ffed004 3fff3f34  
3fffeed0:  3fff41a8 3fff80ea 3fff80cc 4028552d  
3fffeee0:  00000014 3fff3f34 00000000 4027ce8d  
3fffeef0:  00000000 0000054a 3ffe899c 40101bc2  
3fffef00:  402a234c 3fff6e2c 3fff47bc 3fff6724  
3fffef10:  3fff3f34 00000008 3fff80cc 4027c551  
3fffef20:  3fffdc80 3fff47bc 3fff6e2c 4027c360  
3fffef30:  402a550d 3fff47bc 3fff6e2c 402a551f  
3fffef40:  3fff80dc 3fff80cc 00000000 3fffdcb0  
3fffef50:  3ffeffa8 00000000 3fff6e2c 402a67fb  
3fffef60:  40000f49 3fffdab0 ffffff01 40000f49  
3fffef70:  40000e19 000d6d03 bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000ed 40106b39  
3fffef90:  40106b3f bff00000 0000bfff 61430065  
3fffefa0:  4010000d bff00000 000d6d03 401000ab  
3fffefb0:  40296eec 3fffef4c 40296e9d 3ffffee8  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
[... 337 lines cut ...]
0x401037a4 in pp_post at ??:?
0x40106a3b in lmacRxDone at ??:?
0x4010432f in rcReachRetryLimit at ??:?
0x4021e5e1 in CRGBPalette16::loadDynamicGradientPalette(unsigned char const*) at ??:?
0x401072d1 in ets_timer_disarm at ??:?
0x4021ec6c in Segment::virtualLength() const at ??:?
0x40220718 in Segment::loadPalette(CRGBPalette16&, unsigned char) at ??:?
0x4022074d in Segment::currentPalette(CRGBPalette16&, unsigned char) at ??:?
0x40107f94 in BusManager::setPixelColor(unsigned short, unsigned int) at ??:?
0x40100a98 in Segment::setPixelColor(int, unsigned int) at ??:?
0x402632d4 in HardwareSerial::available() at ??:?
0x4020ac9c in mode_noise16_4() at ??:?
0x40266d46 in yield at ??:?
0x402209cc in WS2812FX::service() at ??:?
0x402a8466 in wifi_get_opmode at ??:?
0x402a964b in wifi_station_get_connect_status at ??:?
0x402397e8 in handleNightlight() at ??:?
0x40266d46 in yield at ??:?
0x40248bf3 in WLED::loop() at ??:?
0x4026ed8c in StreamNull::~StreamNull() at ??:?
0x402687cc in uart_write at ??:?
0x40263398 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

andrehen, 2024-01-26:
More information: 

I did the crash scenario 3 times. And even though the crash happened while oppening the page "LED Preferences", the stack trace is different:

Stack 1:
```
0x40101f44 in malloc at ??:?
0x402a5291 in pbuf_alloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:669
0x4028a940 in ieee80211_deliver_data at ??:?
0x4028f0b0 in sta_input at ??:?
0x402a26b7 in pp_tx_idle_timeout at ??:?
0x402a1f7b in ppPeocessRxPktHdr at ??:?
0x402a67fb in ets_snprintf at ??:?
0x40106b39 in call_user_start_local at ??:?
0x40106b3f in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
0x401000ab in app_entry_redefinable at ??:?
0x40296eec in cont_ret at cont.S.o:?
0x40296e9d in cont_continue at cont.S.o:?
0x40107e8c in spi_flash_read at ??:?
0x40262e5c in EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at ??:?
0x40262ee9 in EspClass::flashRead(unsigned int, unsigned char*, unsigned int) at ??:?
0x402755f9 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:196 (discriminator 1)
0x40266d46 in yield at ??:?
0x40267fb8 in flash_hal_read at ??:?
0x4025435e in littlefs_impl::LittleFSImpl::lfs_flash_read(lfs_config const*, unsigned int, unsigned int, void*, unsigned int) at ??:?
0x402012a5 in lfs_bd_read at lfs.c:?
0x40254340 in littlefs_impl::LittleFSImpl::lfs_flash_read(lfs_config const*, unsigned int, unsigned int, void*, unsigned int) at ??:?
0x4020215e in lfs_dir_find_match at lfs.c:?
0x40201886 in lfs_dir_fetchmatch at lfs.c:?
0x40201bda in lfs_dir_find at lfs.c:?
0x40202134 in lfs_dir_find_match at lfs.c:?
0x402759e4 in _vsprintf_r at /workdir/repo/newlib/newlib/libc/stdio/vsprintf.c:63
0x40204acb in lfs_stat at ??:?
0x402540bc in littlefs_impl::LittleFSImpl::exists(char const*) at ??:?
0x40221aee in WS2812FX::loadCustomPalettes() at ??:?
0x401011a0 in ets_post at ??:?
0x401072d1 in ets_timer_disarm at ??:?
0x402747b8 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x4025f758 in esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at ??:?
0x40101f44 in malloc at ??:?
0x40266ad3 in operator new(unsigned int) at ??:?
0x40260329 in esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at ??:?
0x4026062e in esp8266::MDNSImplementation::MDNSResponder::_writeMDNSRRDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRDomain const&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at ??:?
0x402607e8 in esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at ??:?
0x40105841 in lmacIsIdle at ??:?
0x40106989 in lmacTxFrame at ??:?
0x40260356 in esp8266::MDNSImplementation::MDNSResponder::_udpAppend8(unsigned char) at ??:?
0x4010656e in lmacMSDUAged at ??:?
0x401011a0 in ets_post at ??:?
0x40103bd5 in rcUpdateTxDone at ??:?
0x401037a4 in pp_post at ??:?
0x40106a23 in lmacTxFrame at ??:?
0x401037a4 in pp_post at ??:?
0x401011a0 in ets_post at ??:?
0x40103bd5 in rcUpdateTxDone at ??:?
0x40105bfe in lmacRecycleMPDU at ??:?
0x401037a4 in pp_post at ??:?
0x40106a23 in lmacTxFrame at ??:?
0x40105bfe in lmacRecycleMPDU at ??:?
0x401011a0 in ets_post at ??:?
0x4010606f in lmacRecycleMPDU at ??:?
0x401037a4 in pp_post at ??:?
0x40106a23 in lmacTxFrame at ??:?
0x40105b37 in lmacProcessTxSuccess at ??:?
0x401037a4 in pp_post at ??:?
0x40104a00 in wDev_ProcessFiq at ??:?
0x40106989 in lmacTxFrame at ??:?
0x401011a0 in ets_post at ??:?
0x40103bd5 in rcUpdateTxDone at ??:?
[... 242 lines cut ...]
0x40223356 in _ZN14NeoEsp8266UartI12UartFeature121NeoEsp8266UartContextE10UpdateUartEb$isra$0 at bus_manager.cpp:?
0x40225ed5 in _ZN7PolyBus4showEPvhb$isra$0 at bus_manager.cpp:?
0x401076bc in system_get_time at ??:?
0x40103bd5 in rcUpdateTxDone at ??:?
0x401037a4 in pp_post at ??:?
0x40106a23 in lmacTxFrame at ??:?
0x40105bfe in lmacRecycleMPDU at ??:?
0x4022074d in Segment::currentPalette(CRGBPalette16&, unsigned char) at ??:?
0x40107f94 in BusManager::setPixelColor(unsigned short, unsigned int) at ??:?
0x40224989 in BusDigital::getPixelColor(unsigned short) at ??:?
0x40225a44 in _ZN24NeoEsp8266UartMethodBaseI26NeoEsp8266UartSpeedWs2812x14NeoEsp8266UartI12UartFeature121NeoEsp8266UartContextE25NeoEsp8266UartNotInvertedE6UpdateEb$isra$0 at bus_manager.cpp:?
0x40225ed5 in _ZN7PolyBus4showEPvhb$isra$0 at bus_manager.cpp:?
0x4021f6e6 in WS2812FX::estimateCurrentAndLimitBri() at ??:?
0x402209d8 in WS2812FX::service() at ??:?
0x402a8466 in wifi_get_opmode at ??:?
0x402a964b in wifi_station_get_connect_status at ??:?
0x402397e8 in handleNightlight() at ??:?
0x40266d46 in yield at ??:?
0x40248c20 in WLED::loop() at ??:?
0x4026ed8c in StreamNull::~StreamNull() at ??:?
0x402687cc in uart_write at ??:?
0x40263398 in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
```

The crash really seens to be related to memory, as malloc is aways the last function called... But why only with iPhone's Safari  this happens? There are features in the code I can disable to release memory and test?

willmmiles, 2024-01-27:
I can replicate this issue with 0.14.1 and Safari shortly after a board reset.  Later requests do not reliably hit the bug, but with repeated requests it will eventually fail.

I'm fairly confident the root issue is the same AsyncWebServer memory corruption problem I've been chasing for a bit now.  Safari just happens to both (a) be fairly aggressive with parallel web requests, and (b) send the right set of headers to trigger the underlying fault.

casesolved-co-uk, 2024-02-08:
@andrehen @willmmiles I had to disable websockets to get v14.1 stable with `WLED_DISABLE_WEBSOCKETS` - should improve things

blazoncek, 2024-02-08:
Websockets, if used by external devices, will in fact contribute to a more stable device.

Unfortunately they do need some RAM and, if that is scarce, it may help to disable them in certain circumstances.

Disabling them has no other disadvantages but possible stale info in UI.

casesolved-co-uk, 2024-02-08:
> Websockets, if used by external devices, will in fact contribute to a more stable device.

You can try to deny my experience if you want, but this is not true.

blazoncek, 2024-03-04:
Anyone interested I posted binaries of 0.14.2-b2, which is intended to solve this issue, on Discord, #beta-testing channel.
It includes AWS modifications from @willmmiles and faster CPU clock.

closed 2024-07-09 by github-actions[bot]