wled/WLED #3727: ESP32 With WLED (Any version) not picking wifi

Junicavi, 2024-01-26:
### What happened?

I wa setting up a new ESP32 board with WLED, and after the install was completed, when I was going to set up my WIFI info, nothing is coming up in the dropdown, I already tried to manually enter the credentials, and nothing.

[image]

### To Reproduce Bug

Install WLED on an ESP32 Board

### Expected Behavior

Be able to configure Wifi

### Install Method

Binary from WLED.me

### What version of WLED?

Any

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

roninniagara, 2024-01-26:
try 13.3 instead of 14.1?

blazoncek, 2024-01-26:
Or use AP mode and configure from that.

closed 2024-06-19 by github-actions[bot]