wled/WLED #3733: Esp hang up after several minutes

madspeed, 2024-01-31:
### What happened?

ESP32 wroom board hangs up after several minutes of work. I am using a 7A power supply and 274 diods (ws2812b).
Firmware is 0.14.1
Not only wifi disappears but also effects freeze.

### To Reproduce Bug

Happens every time after boot

### Expected Behavior

Stable work

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-31:
Please obtain debug build and monitor serial output.
Debug builds can be found in @srg74 WLED-shield repository.
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Latest/Debug_builds

madspeed, 2024-01-31:
is there any manual how to monitor serial output?

blazoncek, 2024-01-31:
If you have Arduino IDE or Visual Studio with PlatformIO then you can use those but any serial tool (like terminal) will do.
Once you have debug build and serial monitor up and running, you will see WLED display periodic messages that may help pinpoint where the error occurs.

Please use WLED [forum](https://wled.discourse.group) or Discord for help and support.

mindstorms6, 2024-02-01:
I've also noticed a regression on 14.1. I have 11 WLED Devices and all of them eventually hang on 14.1. Lose wifi and only solution is to reboot. Works for a short while but all get into a bad state again. Flashing to 14.0 seems to fix the issue.

madspeed, 2024-02-02:
You were right, 14.0 solved the problem!
Thank you.

doncalitri, 2024-02-04:
Having the same issue since 0.14.1. 
need to completely cut power but after a short while both esp32 with wled version 0.14.1 is not reachable via WiFi anymore and led animations are stuck/frozen.  No option to downgrade for me as the boards are in the ceiling.  Hopefully fixed in next release on short bases.

madspeed, 2024-02-04:
I was able to downgrade by wifi. Just switch off the light after the boot of esp32. And start flashing process as soon as possible.

blazoncek, 2024-02-13:
Please try 0.14.2-b1 and report back.

nonojoeli, 2024-02-24:
I experienced the same hangups on 4 different devices with 0.14.1 and also 0.14.2-b1. 
After changing back to 0.14.0 all is stable again.

nonojoeli, 2024-02-24:
I installed 0.14.2-b1 with debug stuff on one of the devices and induced a crash. The log file is attached:

[wled-0.14.2-b1-001.log](https://github.com/Aircoookie/WLED/files/14394353/wled-0.14.2-b1-001.log)

Some info:
mqtt publishing on
multiple commands via node-red-contrib-wled node 
 -> it does a sequence of psave, some segment flashing and a last ps

If I can help with more logging/testing I'm up for the task.

blazoncek, 2024-02-25:
@nonojoeli there is no crash in the log.

nonojoeli, 2024-02-25:
This might be the case - it still prints out the regular DEBUG INFO blocks.
But the device is completely stuck - effects are stopping and network is unresponsive (at least HTTP, did not check pings).

blazoncek, 2024-02-25:
you will need to provide your WLED configuration (cfg.json) and presets (presets.json)

efaden, 2024-03-17:
I'm also experiencing this with the same board.  Was there ever a resolution?

blazoncek, 2024-06-26:
Closing as no response received.

closed 2024-06-26 by blazoncek