wled/WLED #3735: Seqment Sync is broken when switching from 0.14.0 to 0.14.1

IIIDPGIII, 2024-02-01:
### What happened?

I have two WLED ESP32 systems each controlling two LED strips each. I use one as a master. On 0.14.0, presets can turn on and off segments 0 and 1 independently and sync to the other WLED to match. Starting with 0.14.1, the segment on/off switch and control only works on the actual device, but not the synced devices. I confirmed & reproduced this by simply switching between 0.14.0 and 0.14.1 on both devices and seeing it start working again on 0.14.0.

Configuration
- WLED-left-master 
    - (seq 0, led 0-1) WHITE LED Strip
    - (seq 1, led 1-60) RGBW LED Strip
- WLED-right
   - Same configuration as above

### To Reproduce Bug

1. Setup two WLED ESP32 devices with two identical segments 0 & 1 configurations each
2. Make sure both WLED ESP32 devices are using 0.14.0 firmware
3. On the master WLED device create a preset that turns on only segment 0 and applies an effect. PRESET "Seqment 0 only"
4. On the master WLED device create a preset that turns on only segment 1 and applies an effect. PRESET "Seqment 1 only"
5. On master device: enable SYNC
6. On master device: Switch between PRESET "Segment 0 only" and PRESET "Seqment 1 only"
7. Test that both WLED devices are correctly switching the segment lights
8. Update firmware on both WLED devices to 0.14.1
9. Rerun Test steps descriped above in step 7.  The LEDs on segment 0 of the device receiving the sync preset should be failing to apply the settings.

### Expected Behavior

The LEDs on segment 0 of the device receiving the sync preset should be failing to apply the settings.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Likely related to bug report https://github.com/Aircoookie/WLED/issues/3701

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

IIIDPGIII, 2024-02-01:
> Could you post what sort of LED strips you're using please? And if possible and you have some available, try some APA102 clock & data strips on segment 0. This seems to improve/fix things.

*Could you post what sort of LED strips you're using please? 
Using on:
     * GPIO 16 / Sequence 0 -  [Dig2Analog ](https://quinled.info/quinled-dig2analog/) (board) that mimics a single SK6812 RGBW LED to control PWM Analog White light strip
     * GPIO 3 / Sequence 1 - SK6812 RGBW 59 LED strip

*And if possible and you have some available, try some APA102 clock & data strips on segment 0.
Hi, I don't have APA102 LEDs

Is there a way to dump this configuration information from WLED?

Here are the JSON files I was able to export
[wled_presets_WLED-left-master.json](https://drive.google.com/file/d/14_vMNkPrYFUxjSbTENWQLAW33qFq4kxZ/view?usp=sharing)
[wled_presets_WLED-right.json](https://drive.google.com/file/d/1lJiHHteyOogEMmXRfBwb0R_qfBxVPtYe/view?usp=sharing)

casesolved-co-uk, 2024-02-01:
@IIIDPGIII Could you try deleting and recreating your presets on v14.1? It looks like there are two separate issues here.

closed 2024-06-08 by github-actions[bot]