wled/WLED #3736: Pushed Pixel Art not being stored in a Preset

DarrenRatcliffe, 2024-02-01:
### What happened?

Pixel art sends image to my 32x32 2D matrix fine but when I create a Present it doesn't seem to store it and when I click on the preset nothing seems to happen. All other existing presents work fine.

### To Reproduce Bug

Follow approach 3 from here https://kno.wled.ge/features/pixel-art-converter/#usage using v1.0.8 (stable) of Pixel Art Converter from here https://github.com/werkstrom/WLED-PixelArtConverter/?tab=readme-ov-file. Load the example WLED logo (scale to 32x32). Send to device (segment 0) save as new Preset, change to other Preset then go back to Preset with Pixel Art and it doesn't work

### Expected Behavior

I expect the Pixel Art that was sent and saved in a Preset to load when Preset is selected

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-01:
Most likely a limit of how many characters there can be in a request.
Tagging @ajotanc since he is the author.

AdamK-DE, 2024-02-09:
pixels in presents is working with v15b but ONLY without segments! .. 
when you use segments then WLED is storing only the segment stuff..

but it works! with ioBroker/Blockly i send API calls that 
1. create a matrix with 14 segments and 
2. fill 5 of the segments with pixels (its a clock like "00:00" for 8x32 Matrix)

sending segments with pixels works but the first API POST send is only creating the segments. The second is filling..
seems to be a bug in wled v15b.. after 1x send it works without 2 times.. until a other preset is changing the segments.

code: http://kopacz.de/blockly wled clock

syntax that works:

`curl -X POST "http://192.168.178.100/json/state" -d '{"on":true,"bri":10,"transition":10,"mainseg":0,"seg":[{"id":0,"start":0,"stop":32,"startY":0,"stopY":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"LinieOben","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":2,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":0,"stop":32,"startY":7,"stopY":8,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"LinieUnten 2","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":2,"start":0,"stop":32,"startY":6,"stopY":7,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"LinieUnten 1","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":11,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":3,"start":0,"stop":7,"startY":1,"stopY":5,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":4,"start":7,"stop":10,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255 ,"cct":127,"set":0,"n":"Zahl 1","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0, "i":[0,2,"000000","FFFFFF",3,5,"000000","FFFFFF",6,8,"000000","FFFFFF",9,11,"000000","FFFFFF",12,14,"000000","FFFFFF"]},{"id":5,"start":10,"stop":11,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":6,"start":11,"stop":14,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"Zahl 2","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0,"i":[0,4,"FFFFFF","000000",5,7,"FFFFFF","000000",8,10,"FFFFFF","000000",11,15,"FFFFFF"]},{"id":7,"start":14,"stop":15,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":8,"start":15,"stop":16,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"Doppelpunkt","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0,"i":["000000","AAAAAA","000000","AAAAAA","000000"]},{"id":9,"start":16,"stop":17,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":10,"start":17,"stop":20,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"Zahl 3","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0,"i":[0,4,"FFFFFF","000000",5,7,"FFFFFF","000000",8,10,"FFFFFF","000000",11,15,"FFFFFF"]},{"id":11,"start":20,"stop":21,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":12,"start":21,"stop":24,"startY":1,"stopY":6,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"Zahl 4","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":f [... cut ...] rue,"frz":false,"bri":255,"cct":127,"set":0,"n":"leer","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}' -H "Content-Type: application/json"`

closed 2024-06-15 by github-actions[bot]