wled/WLED #3739: Potentiometer lag ESP8266

AudioCrafter, 2024-02-03:
### What happened?

I have a ESP8266-12F (d1 mini), connected to a 10k potentiometer, set up to modify the master brightness. (issue also while modifying other settings like segment brightness or hue)
When i rotate the potentiometer, the brightness is stuttery / lagging behind extremely. If I dim the brightness via the webui, it is smooth as expected.
I tested it with 4 differernt ESPs and potentiometers, so i think it is a bug?

https://github.com/Aircoookie/WLED/assets/70272122/27868120-893e-4795-8c5a-d2b6adfbf582

### To Reproduce Bug

Connect:
ESP Gnd -> Potentiometer pin1
ESP A0 -> Potentiometer pin 2 (middle pin)
ESP 3.3v -> Potentiometer pin3
(ESP D4 -> LED Stripe Data in, but i think thats irelevant to reproduce this behavior)

WLED Settings > LED Preferences > Button 1 GPIO: Analog 

Also here my complete config:
[wled_cfg_wohnraum.json](https://github.com/Aircoookie/WLED/files/14151816/wled_cfg_wohnraum.json)
[wled_presets_wohnraum.json](https://github.com/Aircoookie/WLED/files/14151817/wled_presets_wohnraum.json)

### Expected Behavior

Smooth dimming, like from the webui 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED Release 0.14.1 Hoshi

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-03:
This is intentional. Potentiometer is read 4/s (every 250ms).

AudioCrafter, 2024-02-03:
Okay, thank you for the quick response :)
What is the reason that it is only read every 250ms? and can i somehow change it easily? 
Thanks again :)

blazoncek, 2024-02-03:
you can, by modifying source (button.cpp)

AudioCrafter, 2024-02-03:
Okay Thank you :)

softhack007, 2024-02-03:
> What is the reason that it is only read every 250ms? and can i somehow change it easily?

There are a few reasons for not reading the pot too often:
* ADC on espressif chips is very noisy. If we read too often and the pot controls brightness, this will lead to flickering
* potentiometers are not joysticks ;-) usually its enough to read them 4  times per second.
* analogread() on 8266 causes a delay of 5-10 milliseconds. If you do that too often, the wifi driver might die (crash).

If you want to explore what happens, you should start with modifying these lines:
https://github.com/Aircoookie/WLED/blob/3d6f25a9d38eb0e6f07c8e81bd9814804d85c47f/wled00/button.cpp#L233

https://github.com/Aircoookie/WLED/blob/3d6f25a9d38eb0e6f07c8e81bd9814804d85c47f/wled00/button.cpp#L145-L148

softhack007, 2024-02-03:
@blazoncek the button / potentiometer code currently does not trigger smooth transitions of brightness, is there anything we could improve? 

Would be nice if analog input causes the same smooth transitions as the UI global brightness control.

https://github.com/Aircoookie/WLED/blob/3d6f25a9d38eb0e6f07c8e81bd9814804d85c47f/wled00/button.cpp#L192

https://github.com/Aircoookie/WLED/blob/3d6f25a9d38eb0e6f07c8e81bd9814804d85c47f/wled00/button.cpp#L216-L217

https://github.com/Aircoookie/WLED/blob/3d6f25a9d38eb0e6f07c8e81bd9814804d85c47f/wled00/button.cpp#L224

blazoncek, 2024-02-03:
> Would be nice if analog input causes the same smooth transitions as the UI global brightness control.

Perhaps adding:
```c++
...
      // global brightness
      briOld = bri;
      if (aRead == 0) {
...
```
will do it.

AudioCrafter, 2024-02-03:
> If you want to explore what happens, you should start with modifying these lines:

I already made a [small fork](https://github.com/AudioCrafter/WLED_smoothpoti) with the potentiometer read time turned down to 50ms, and so far it worked very well for me :)

Thank you very much for the great support and question answering, couldnt have asked for more :)

closed 2024-06-19 by github-actions[bot]