wled/WLED #3744: Preset feature not working on mobile application

jotta237, 2024-02-05:
### What happened?

The preset feature on the mobile app is not working. I press my preset like i would do through the Web, but it won't work. 
Idk why it could be or if it happens to everyone. Could be a WiFi problem, idk.
I just wanted for the community to know it.

### To Reproduce Bug

Create a preset and try to use it on the mobile application

### Expected Behavior

The preset should activate as it does when using it on the web

### Install Method

Binary from WLED.me

### What version of WLED?

WLED Mobile 1.1.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

austinhutchen, 2024-02-06:
Same issue

jotta237, 2024-02-06:
> Please open an issue for the mobile app - it has its own repo here https://github.com/Aircoookie/WLED-App/issues
> 
> It would help if you can identify your phone or tablet model, android version, and provide a step-by-step description/screenshots of what you try to do. Also state if you're using AP mode or a regular wifi router connection.
> 
> The mobile app is working in my case, so it could be something specific to your device or setup.

Thanks for the comment. I'll open an issue on the app repo as soon as possible with screenshots and my phone specifications.

blazoncek, 2024-02-06:
I would recommend to use WLED-Native app by @Moustachauve as official WLED app is unmaintained currently.

closed 2024-02-06 by blazoncek