wled/WLED #3745: ESP32 Sync is not working

robert-ke, 2024-02-06:
### What happened?

I don't found the Problem, i have two ESP32 D1 Mini NodeMCU Boards - WLED Version 0.14.1-b3 - everything is fine and works.
I activate the Sync Button on both, but it is not working. The InstanceList is empty, but Settings are correct.

I tested it with Firmware 0.13.3 and higher - Sync will not working on the ESP32 Boards. I have 3 new Boards:
https://www.amazon.de/dp/B08BTLYSTM?psc=1&ref=ppx_yo2ov_dt_b_product_details

... but the same Problem.

On the ESP8266 D1 Mini Boards Sync works fine.

Bug?

Here are some Screenshots:
[image]

[image]
and the Settings:
[image]

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-06:
Only one needs sync button to be active. Others listen automatically.
Make sure your ports are the same on all devices and sync groups are properly selected.

robert-ke, 2024-02-06:
yes, i think the Settings are correct:
[image]

and the Sync Button is on the First Controller Active:
[image]

If i change a color, the ESP8266 Board change it correct - but the second ESP32 Board - Here WLED-2 not.

Any other Idea?

blazoncek, 2024-02-06:
If you Instances dialog is empty, check your WiFi settings. The devices cannot *see* each other.
Most likely you have some sort of "client isolation" enabled.

robert-ke, 2024-02-06:
I don't think that's the problem. 
Because the sync runs under the D1 Mini with ESP8266 in the same network/WLAN. 
I'll keep testing, thank you very much

closed 2024-02-06 by robert-ke