wled/WLED #3749: WLED crashing on 0.14.1

guyman575, 2024-02-07:
### What happened?

Added a new WLED node to my home assistant, and was setting up some automation. When I did this I upgraded both to the most recent, but I seem to have isolated the problem to this particular version since it doesnt occur on 0.14.0.

FWIW I am controlling the lights via pyscript home assistant integration
https://hacs-pyscript.readthedocs.io/en/latest/overview.html

I haven't had an issue with this though until 0.14.1.

### To Reproduce Bug

Running a script that send multiple concurrent WLED requests to change light colors. Results in crashing nondeterministically 

### Expected Behavior

expect lights to function as they do in 0.14.0

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
2024-02-06 22:24:11.313 ERROR (MainThread) [homeassistant.core] Error executing service: <ServiceCall light.turn_on (c:01HP13S4Y8ATCQ63F54DCD6BBE): entity_id=['light.shoe_shelf_5'], params=rgb_color=(255, 0, 255), brightness=255>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 188, in request
    response = await self.session.request(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 574, in _request
    conn = await self._connector.connect(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/connector.py", line 544, in connect
    proto = await self._create_connection(req, traces, timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/connector.py", line 911, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/connector.py", line 1204, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/connector.py", line 992, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1069, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 628, in sock_connect
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 187, in request
    async with async_timeout.timeout(self.request_timeout):
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 28, in handler
    await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/wled/light.py", line 251, in async_turn_on
    await self.coordinator.wled.segment(**data)
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 515, in segment
    await self.request("/json/state", method="POST", data=state)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 224, in request
    raise WLEDConnectionTimeoutError(msg) from exception
wled.exceptions.WLEDConnectionTimeoutError: Timeout occurred while connecting to WLED device at 192.168.86.37

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 2083, in _run_service_call_catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 878, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 580, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 34, in handler
    raise HomeAssistantError("Error communicating with WLED API") from error
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-07:
Please provide WLED crash dump (from debug build) as HomeAssistant logs are useless.

blazoncek, 2024-02-13:
Please try 0.14.2-b2 and report back.

idiotsguide, 2024-02-15:
> Please try 0.14.2-b2 and report back.

FYI, I am using Athom WLED bulbs and had the same issue as the original post. Updating to 0.14.2-b2 as you suggested seems to have solved the issue for me. Thank you!

klaneos, 2024-02-20:
My issue is more or less the same so I’ll place here rather than creating a new issue.

WLED 0.14.1 sound reactive
QuinLED-Dig-Quad board with QuinLED-ESP32-AE+ (for mic input)
24vdc power supply and 24vdc LEDS with surplus power to run everything on max brightness.
RGBW LED strip (requires swap of G&W LEDs for correct order.
Cables sized appropriately.
Five segments in presets.
Calling presets from WLED app is fine.
Calling presets from HA (in the form of scenes) crashes ESP32. Set up a HA controlled outlet to cycle power supply when ESP32 has crashed. Not ideal as lights go off when cycled.
Have tried 2 ESP32’s ( standard with dig-quad and AE+mic) with the same result.
Was able to call presets from HA on 0.14.0. Crashed only once in the past. However 0.14.1 crashes 100% of the time when calling presets.
My temporary work around is to call WLED power ON from HA, with brightness adjusted to suit the time of the evening. It will maintain the last preset when switched on. This keeps the lights switching on and off and works flawlessly, however, it’s not enough to perform complex automations with presets.
Happy to discuss and compare issues to provide as much info to get a fix in place. Happy to crash the ESP32 again and grab the WLED log if need be.
Also tried 0.14.2-b2 via WiFi load with the same result.

blazoncek, 2024-02-21:
Most likely duplicate of #3685
Closing

closed 2024-02-21 by blazoncek