wled/WLED #3751: Light glitch flashes regularly

cdalexndr, 2024-02-07:
### What happened?

Every few seconds all LEDs flash quickly, see video:

https://github.com/Aircoookie/WLED/assets/22418250/7e65c7da-de55-4553-81a5-6b779c13ad3e

### To Reproduce Bug

Solid color, notice flash.

### Expected Behavior

No flashing

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Issue not reproductible with WLED sound reactive 0.13.4, so this is **not a hardware issue**.
https://github.com/wled-install/wled-install.github.io/blob/main/bin_dir/2_Sound_reactive_version/0.13.4/WLEDSR_0.13.4_esp32_4MB_M.bin

WS2815 leds:
[image]
[cfg.json](https://github.com/Aircoookie/WLED/files/14192979/cfg.json)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-07:
HomeAssistant? MQTT? Alexa? Hue?
Any of those? If yes, then try to disable and retry.

cdalexndr, 2024-02-07:
After more testing, the glitch also happens with 0.13.4, but more rarely.

Indeed, HomeAsssistant WLED Integration was the culprit.
Disabling "Enable polling for updates" solved this issue.
[image]

HA wled integration diagnostics: 
[config_entry-wled-6c600e5f3b03226e216b08a97e0179ca.json.txt](https://github.com/Aircoookie/WLED/files/14201800/config_entry-wled-6c600e5f3b03226e216b08a97e0179ca.json.txt)

blazoncek, 2024-02-07:
Instead of polling ask HA developers to subscribe to websockets for updates.

cdalexndr, 2024-02-07:
Opened [issue on HA ](https://github.com/home-assistant/core/issues/109914) on HA side

Panoramiac, 2024-02-07:
I might have the same issue here. In my case the end of a 364 SK6812 RGBW LED strip from BTF-LIGHTING  (about 12m) starts to flicker heavy from time to time. I now disabled the polling in HA. I will observe it and provide and update (including a video). It is powered by a 30A power supply with separate power injections and works since 2 years, so I was thinking more about a SW bug instead of a hardware issue as it was not touched since then.

jnpatrick99, 2024-02-07:
This bug has been in WLED 0.13+ firmware for quite some time. Personally I gave up and stopped using WLED because if it a year ago, and just bought a finalized HW/SW solution from 3rd party on Amazon.

DynamoBen, 2024-02-08:
I have the same issue with the releaseed version on on set of Eufy LEDs which use SK6812 RGBW. I noticed this issue during the betas, to correct it I had to load the previous version, factory reset, and recreate all the presets. To troubleshoot I deselected everything in `Sync Interfaces` and deleted the device from HA, still had the flicker. I am using the QuinLED-Dig-Uno with Ethernet if that matters but only running WiFi. Happy to try other things if there are any suggestions.

blazoncek, 2024-02-08:
The issue comes (as it was written on many occasions) from the resource exhaustion on WLED side. When too many HTTP requests are made ESP cannot handle everything in a manner that would feed LED data consistently (with everything WLED has to offer).
The longer the LED strip the more pronounced or frequent the flashes.

LED data is fed by NeoPixelBus (the underlying driver) and there is little WLED can do.

You can help a little by flashing a firmware with faster CPU and/or flash clock but not all ESPs will handle that gracefully.

Panoramiac, 2024-02-08:
> The issue comes (as it was written on many occasions) from the resource exhaustion on WLED side. When too many HTTP requests are made ESP cannot handle everything in a manner that would feed LED [...]

Interesting, are these overclocked images available or would we need to built them by our own? For now I disabled the polling for now and could survive with it. Higher CPU clock might also require a passive cooling element on the chip.

blazoncek, 2024-02-08:
At most times faster flash speed helps. I never used CPU overclocking though.

Add this to your PIO environment when building:
`board_build.f_flash = 80000000L`

Akaricchi, 2024-02-09:
See also #3592. I have posted a workaround patch for the HA polling issue there.

blazoncek, 2024-02-09:
> See also #3592. I have posted a workaround patch for the HA polling issue there.

But you never cared to make a PR so anyone could benefit.

Akaricchi, 2024-02-09:
> > See also #3592. I have posted a workaround patch for the HA polling issue there.
> 
> But you never cared to make a PR so anyone could benefit.

Hey, at least I cared to publish my findings before I find the time to make a more robust PR that's actually suitable for merging.

closed 2024-06-17 by github-actions[bot]

wizbowes, 2025-12-14:
0.15.3 is flashing for me.  HA all updated.  All upgraded and flashing so ended up here.  I have disabled the polling and it seems to have fixed it