wled/WLED #3753: Mac

pr8x, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
Would be nice if I could somehow change the mac address via WLED

**Describe the solution you'd like**
A simple setting for setting the MAC.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
I want to target different ESP32s using a custom ESP-NOW remote.

Thank you for your ideas for making WLED better!

softhack007, 2024-02-10:
There is an API from espressif for setting the MAC address, with some examples here:
https://esp32tutorials.com/esp32-esp-idf-get-set-custom-mac-address/

However I'm not really a friend of this idea. 

The use cases are very special, and every day "Joe Random" will probably never want to modify the MAC. 
On the other hand, you can render your device (and even the wifi network) unusable if you make mistakes. The MAC is expected to be unique, so strange things will happen if two boards on the same wifi use the same MAC.

pr8x, 2024-02-11:
I agree, I will probably just write a usermod for this