wled/WLED #3761: wled suddenly stops working

papinist, 2024-02-15:
### What happened?

no more connection to web ui, no response

### To Reproduce Bug

I don't know.. I din't anything new

### Expected Behavior

it should work as always

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1, also 0.14.2b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
{l $ܟ|  l | l c|    ; b c  on l'g   cp  ls${l8 '  $  c' <   b  'g  $  d` onl  '; ۛgb$  $x g s  ܜ  c' < c  go  d` n'$` g{   o#  ` ;  o#  `  Ǉd`  o l    ' r  o| d l c  <{ d '  g  l`  s $ $  Ada

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4028faa7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebe0 end: 3fffffb0 offset: 0150
3fffed30:  40289cd0 3fff6268 3fff69d4 3fff6268  
3fffed40:  3fff6bf4 40289ed3 3ffef748 3fff426c  
3fffed50:  00000000 00000002 00000001 3ffee2b0  
3fffed60:  3fff6c3a 4010707b 3fff6268 3fff3c74  
3fffed70:  3fff60f4 3fff65f0 3fff3c8c 40282188  
3fffed80:  00000002 4027a68d 3fff6268 3fff3c74  
3fffed90:  00002990 00000532 3ffe899c 40101bc2  
3fffeda0:  00000016 3ffef748 00000020 3fff3cd8  
3fffedb0:  3fff3cd8 3fff6bf4 3fff6bf4 4027a8bb  
3fffedc0:  3fff3d10 3fff6bf4 00000000 4027ad1e  
3fffedd0:  3fffee00 3fffee90 3ffef748 00000016  
3fffede0:  3ffeefa0 3fffee90 3ffef748 3fff6bf4  
3fffedf0:  3fff6bf4 3fff6378 3fff3cd8 402823a1  
3fffee00:  005e0001 3ffffb00 3fffee90 00040000  
3fffee10:  3fff6c48 3fff3cd8 3fff6378 40283e8c  
3fffee20:  40104a00 00080000 00000000 3fff3cd8  
3fffee30:  00000018 3fff3cdc 00000001 00000000  
3fffee40:  00000002 3fff40d4 00000004 4027b5cd  
3fffee50:  3ffefd10 3fff4284 00000000 3fff3cd8  
3fffee60:  00000000 3fff6bf4 3fff6c60 40283ef4  
3fffee70:  3fff3cd8 3fffeea0 00000004 3fff3cd8  
3fffee80:  3fff6374 3fff40c8 3fff6374 40282815  
3fffee90:  3fff3cd8 3fffeea0 00000004 402c6880  
3fffeea0:  00000494 00000000 00000020 40101f44  
3fffeeb0:  3fff6378 00000000 00006108 4027b5cd  
3fffeec0:  000000b0 00000016 3ffe899c 00000002  
3fffeed0:  3fff3cd8 3fff40c8 3fff6374 40282be0  
3fffeee0:  4027fe18 00000020 3fff41e4 402c6888  
3fffeef0:  4027fe18 000060a4 3fff3f84 4027fe28  
3fffef00:  402c6888 00000000 3ffee508 4027b681  
3fffef10:  01798214 3ffefd10 3ffefce8 4027ffb8  
3fffef20:  402a52e0 3ffefce8 3ffefd10 60000600  
[... 268 lines cut ...]
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000d4480
~ld
    '  s  g| l d`b  |s l g  o  $`  r l d  Ada
s$ d  |  l | l c|    { # c  n' $'o   #8Ď$;ds$x g  $  #n <   c  n'  $  d` o'$  '{Ǜ o#d  dp o r ܜ   c' | c  go  d` n'$` g{   o#  `rۛgbě`    $`  g l    ' r  o| d $ c  |{ d '  g  l   s d d  Ada
;l l  <  $ < $ b|    s c #  'g dgo   #8Ď$;ds$x g  l  bo |   #  'g  d  d` n'$  '; ۛgb$  $x g s  ܜ  #o | c  on  d` o'$  '; ۛgbě` {  'c  `  Ǉd`  o l    g ;  '< d $ c  <{ d n  '  l`  { $ l  Ada
```

### Anything else?

I use this wled installation on a wemos d1 mini for years now, using it from home assistant and a motion sensor. I found it like crashed and no connection to HA. Web UI was not reachable, tried to power off and on but no luck.
So I connected my laptop via usb and tried to reflash 0.14.1 with no luck, tried 0.14.2b1 same results, tried to erase user data, reconfiguring wifi, still no luck.
The only info I can get is that log obtained via usb.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-15:
Please use exception decoder with monitoring debug output as it will help diagnosing the crash.

Add `monitor_filters = esp8266_exception_decoder` to PlatformIO environment used for building.

papinist, 2024-02-15:
Sorry but this sound like Klingon to me.. I never "build" anything, just used the binary provided about 4 years ago to flash the d1 mini the first time, and tried the web installer now.
I could try another d1 if there is suspect that it could be an hardware issue.

blazoncek, 2024-02-15:
If you can't speak (or read) Klingon, then I'm afraid I cannot help you. 😉 

Without crash dumps from debug build and exception decoder it's just shooting in pitch black.

papinist, 2024-02-15:
I understand your point of view. but I thought this project was also for regular users and not only for programmers. How could a simple "noob" user knows how to compile something? People who needs help probably don't know much about this :)

I tried this project and it worked for a very long time without issues! Now suddenly and without any change (the last update should be that of mid-january) it won't work anymore..

Is there a place where to find a debug-ready binary?

blazoncek, 2024-02-15:
Anyone using Github is assumed developer of some kind IMO.

If you are a regular/ordinary user then please use WLED [forum](https://wled.discourse.group) or Discord for help and support.

BTW WLED has come a long way in last 2 years adding tonnes of new features. In fact ESP8266 is on its limits as far as running it goes.You may want to consider switching to ESP32.

papinist, 2024-02-16:
Tried to downgrade to 0.14.0 solved all the issue, and my d1 mini came back to life :)

The only info that I can give to all you developers, as a simple user, is that on my wemos d1 mini 0.14.0 works, 0.14.1 after a month stopped working.
I don't use any new features added in the last years so I can stay on this until home assistant is able to communicate..

Thanks for this great project!

closed 2024-02-16 by papinist

coolio75, 2024-02-22:
I have the same problem with a D1 mini. 0.13.1 works fine | 0.14.1 No response.
I have tested this with several D1 mini. The same problem with all of them.  There is definitely a bug here.

bouboun59, 2024-03-06:
I have the same problem since 0.14.1 after 4-5 days wled becomes unvailable ping respond 1/100 no inetrface web, after 3-4 reboot after it's ok, but only one reboot it's no efficient.
I don't understand the problem, i have no log.
So I returned in 0.14.0 ad it's work perfectly.

I have D1 Mini Nodemcu ESP8266

blazoncek, 2024-03-06:
Please try 0.14.2-b2 available on Discord, #beta-testing channel.

bouboun59, 2024-03-06:
I have the same problem issue with 0.14.2-b2 for information i have already test the beta versions.

bouboun59, 2024-03-23:
Hello
what's news ?
For me
24/02 v14.0 no problem
6/03/2024 20H00  v14.1.B2
10/03/2024  same issue that B1 BUG
10/03/2024 10H15 v14.0 no problem since 12days