wled/WLED #3762: Presets fail to save/fail to load messages

mharris1984, 2024-02-16:
### What happened?

When  clearing out old presets and deleting them all to build for new segments and added on LED, after deleting the last preset, anytime you click presets, you get a "sorry there was an issue loading your presets" and a box with what appears to be a JSON config of the last preset that was there.  I can copy to clipboard and click new preeset, and fails.  I save any new presets, it fails.  It says to refresh the page to see the new presets and I do and I get the "failed to load" message again, but this time the JSON is the preset config I just tried to save.  

### To Reproduce Bug

- Delete all previously configured presets/playlists.  
- Attempt to save any new preset or playlist. 
- Error of failed to load and JSON formated in the boxwith copy to clipboard pulls up.  
- No matter what you do, no new presets will save, and continue to error after each reload. 

### Expected Behavior

Expected is preset is expected to save and be selectable like any other preset setup
[image]
. 

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0 & 0.14.1 & 0.14.2-b1

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S3

### Relevant log/trace output

```shell
I can pull any logs needed.
```

### Anything else?

screenshots in the above showing the issue that happens.  I see similar behaviour with segments as well when deleting all to start over... I save new ones and refresh and suddenly old ones appear, or they all show gone, or other odd things every refresh. happends on updated from older version ESP32's, and even on newly flashed ESP32 with 0.14.2b1 from new and no restores done, but built from a fresh install. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mharris1984, 2024-02-18:
This did not resolve it for me.  I even tried this on a newly installed WLED on a new ESP32 and same issue.  When I delete ALL presets, and go to save any new ones, they fail, yet show in the JSON list to "copy to clipboard" like in my screenshots above.  I did the clear local storage and went back and same issue, rebooted, and same issue, cleared, saved, rebooted, and still same issue.  Once all presets are deleted, I can no longer make ANY presets to save.  Is there a file on the ESP i can edit the JSON direct and save it, or delete the corrupted one?

blazoncek, 2024-02-18:
Please try 0.15 and report back. It is available from @srg74 Wemos shield repository.

garnbleiche, 2024-03-25:
Same Problem with self compiled WLED 0.15.0-b1 (build 2403220).
After some editing & deleting all presets and playlists i can't save any new preset. "Clear local storage" doesn't help.
Only deleting the presets.json file helps.

mharris1984, 2024-07-24:
> Same Problem with self compiled WLED 0.15.0-b1 (build 2403220). After some editing & deleting all presets and playlists i can't save any new preset. "Clear local storage" doesn't help. Only deleting the presets.json file helps.

I found this too. I had to delete this file, then manually copy and existing GOOD presets.json file over and edit it via the web interface edit screen.  Once I put on some working preset, it's good. Or if I copy the last known good presets that fails to restore, and do these steps after copying it, and paste into the new presets.json file it works. The data is corrupted so I have to sit through it to figure out what is bad in it, fix it, then it's working. 99% or the time, it's easier to flatten it and reinstall a new wled install! I wish the auto recovery worked or the clear local files and reboot worked.  But now, I copy all configs and settings backups after I build a new controller and save it. So if it corrupts, I just reinstall and restore configs