wled/WLED #3764: New hardware revision of D1 mini has poor wifi connection on any version newer than 0.13.1

Boerny41, 2024-02-17:
### What happened?

I was having problems connecting to the WebUI of newly installed ESPs, and after testing I noticed that the wifi connection had a high latency and would drop every few seconds.

This only happened on new ESPs, not on old ones that were upgraded to >0.13.1.
Looking at the ESPs, there were hardware changes made by the manufacturer. 

To confirm this, I tried two old D1 minis and three new ones.
Both old ones work fine with any version, all three new ones do **not** work with any WLED version greater than 0.13.1. All three have high latency and drop the signal.

Below are two pictures. On top is the new hardware revision. Note the missing capacitors on D1/D2 and the missing component above the USB port.

[image]
[image]

### To Reproduce Bug

Install WLED on a newer D1 mini with the manufacturer's modification shown above.

### Expected Behavior

Newer hardware versions supporting new software versions

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1 working on all | >0.13.1 only working on old hardware

### Which microcontroller/board are you seeing the problem on?

ESP8266, Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

coolio75, 2024-02-22:
I have the same problem with a D1 mini. 0.13.1 works fine | >0.13.1 No response

coolio75, 2024-02-22:
@Boerny41 have you tried 15 beta? [WLED at 0_15](https://github.com/Aircoookie/WLED/tree/0_15)

LuxBibi, 2024-02-25:
I can confirm. Have 3 D1 Mini's ... All with > 0.13.1 is slow, and disconnects ... while 0.13.1. is TOP coverage. Have run a PingInfoView (Windows Tool) to "measure" rate of disconnecgted D1 Mini's ...  Shows in my case < 5% pin loss, but sometimes up to 15 consecutive failed pings. :-(
I'll check what are the differences in between 0.13.1 and new releases, and will probably go for 0.13.1, if this delivers the needed functionalities.

BTW: Superb project, to have all these functionalities available in WLED ! Thumbs up guy(s) .. ;-) .. Really apreciate what you have developped .. 

Greeting from Grand-Duchy of Luxembourg ;-))

DanoneKiD, 2024-03-04:
Same issue here.

blazoncek, 2024-03-04:
FYI: there were no changes to WiFi core between 0.14.0 and 0.14.1 except for a new option to force WiFi Phy G mode instead of default N mode #2922. But 0.14.1 uses more RAM and hence stresses ESP more.

And BTW the picture above shows (cheap? and probably poor quality) Lolin clone.
Get official Lolin Wemos D1 mini or ESP32 if you can.

Boerny41, 2024-03-05:
> And BTW the picture above shows (cheap? and probably poor quality) Lolin clone.

You are right, these are just clones and real Lolin ESPs might work better. But the truth is that most people use clones, so many people will have connectivity problems with versions newer than 0.13.1. 

If the devs decide that WLED is only supported on Lolin boards then we can save some time and close this issue.

blazoncek, 2024-03-05:
We cannot guarantee workings on every hardware even though we try (I myself have 15 batches of approximately 50 ESP8266 devices to test on).
I've mentioned elsewhere that for 0.14 we needed to upgrade ESP8266 core to support new LED types and allow enhanced functionality. If (a particular) hardware has issues with it then it is a problem of the hardware, most likely being out of specs.

You can try 0.14.2-b2 (available only through Discord ATM) to see if your board will boot at all as we increased CPU frequency from 80MHz to 160MHz to give ESP8266 a bit more CPU power. Lower quality ESP8266 (found on cheap clones) will most likely not boot or will operate incorrectly. In this case my only recommendation is to use such devices for something else and not WLED.

Jason2866, 2024-06-27:
Probably the onboard 3.3V LDO is to weak. Clones do use fake LDOs or just 150mA types. Replacing the onboard LDO with a 500mA type (as the orig. Lolin board has) does solve the issue most of the time. All my failing WemosD1 mini clones failing before started to work without issues after replacing the LDO.

blazoncek, 2024-07-26:
One of the possible solutions was mentioned [here](https://wled.discourse.group/t/solved-esp8266-no-webui-with-version-newer-0-14-0/12304/11).

andyb2000, 2024-12-31:
Just to add a me too to this, replaced an older unit and applied 0.15
Got around 80% packet loss and web-gui was unusable. I tried to set disable wifi sleep to check and no difference.
Dropped down to 0.14.0 and immediately ping responses worked fine and no loss, so confirmed this is an 'issue' but also agree that clone hardware looks like an issue with the later esp wifi library.

DedeHai, 2024-12-31:
As mentioned, most of these issues are related to poor power design on these clones. What can help is a large capacitor between 3.3V and GND, 1000uF  or even more help to keep the voltage steady.

mondychan, 2025-01-14:
same, wemos d1 clone, running 0.13.1 everything perfect, 
on 0.15.0 90% packet loss, and those that came trought were in hundreds of miliseconds,

Boerny41, 2025-01-14:
I also got multiple genuine lolin boards by now but they show the same behavior.

It might be the poor design of the cloned boards (and apparently genuine ones as well) or it might be a software problem. After all v.0.13.1 works so I'd guess that something has changed since then.

SHU-red, 2026-09-08:
Hi there, wanted to add up to this.

- I already have set up multiple WEMOS D1 minis (8266) and running 0.15.x or 0.16.x (via update)without problems
- Now tackling my next project i flashed directly 0.16.x via Web Flasher or esptool CLI and massive problems connecting to WebUI
- Testet multiple hardware revisions / USB C/Micro and nothing seems to work anymore
- Successfully flashed 0.14.0 like often proposed and immediately worked as expected

The thing that bothers me is, why flashing directly 0.16.x behaves different than a upgraded device from older versions.
I also upgraded it through HA from 0.14.0 to 0.16.1 again and it became immediately unresponsive again.
But then i power-cycled it and it works fine for now.

Slop Gemini Says something like:
> it could be that coming from a more lightweight 0.14.0 with LittleFS already set up there may be using less resources influencing the behavior in a positive way. (Heap, RAM starvation)

1. Can anyone confirm that the following procedure may enable having up-to-date ESP 8266 devices:
> 1. Flash 0.14.0 and setup Wifi/Settings
> 2. Upgrade to newes release

2. Is there a professional evaluation on the different behavior of `flashing 0.16.x` and `upgrading to 0.16.x`?