wled/WLED #3765: Loxone LX Value in combination with RGBW Stripes.

katzendrama, 2024-02-17:
Hi!

I use loxone for quite a while in combination with RGB stripes like the classic WS2812B. In my new house I want to use them with WS2814 RGBW Stripes. The problem is that the White channel is never used when controlling it via /win&FX=0&SS=3&LX=<v>.

Loxone is generally using RGBW in their own products (for example the Ceiling Light Tree). So its possible to convert the LX to RGBW values without setting the white channel to 0.  Thats kind of bad, because you lose a lot of white brightness, especially when you want a bright light. 

As an alternative I considered to add a function in Loxone config to calculate if I want pure white (when R=G=B) and then turn on the white channel to 100%, but that all is too complicated.

So my solution was to implement this in the lx_parser.cpp.

lx_parser.cpp
`#include "wled.h"

#ifdef WLED_ENABLE_LOXONE

/*
 * Parser for Loxone formats
 */
bool parseLx(int lxValue, byte* rgbw)
{
  DEBUG_PRINT(F("LX: Lox = "));
  DEBUG_PRINTLN(lxValue);

  bool ok = false;
  float lxRed = 0, lxGreen = 0, lxBlue = 0, lxWhite = 0;  // *ADDED BY ME Added lxWhite variable

  if (lxValue < 200000000) {
    // Loxone RGB
    ok = true;
    lxRed = round((lxValue % 1000) * 2.55);
    lxGreen = round(((lxValue / 1000) % 1000) * 2.55);
    lxBlue = round(((lxValue / 1000000) % 1000) * 2.55);

    // *changed nothing here. When the normal RGB-Selector is used in loxone no white should be added, just like Loxone reference
  } else if ((lxValue >= 200000000) && (lxValue <= 201006500)) {
    // Loxone Lumitech
    ok = true;
    float tmpBri = floor((lxValue - 200000000) / 10000); ;
    uint16_t ct = (lxValue - 200000000) - (((uint8_t)tmpBri) * 10000);

    tmpBri *= 2.55f;
    tmpBri = constrain(tmpBri, 0, 255);

    colorKtoRGB(ct, rgbw);
    lxRed = rgbw[0]; lxGreen = rgbw[1]; lxBlue = rgbw[2];

    float white = tmpBri; // *ADDED BY ME to allow manipulation on the value (did /2 for testing and such things)

    lxRed *= (tmpBri/255);
    lxGreen *= (tmpBri/255);
    lxBlue *= (tmpBri/255);
    lxWhite = white * (tmpBri / 255); // *ADDED BY ME
  }

  if (ok) {
    rgbw[0] = (uint8_t) constrain(lxRed, 0, 255);
    rgbw[1] = (uint8_t) constrain(lxGreen, 0, 255);
    rgbw[2] = (uint8_t) constrain(lxBlue, 0, 255);
    rgbw[3] = (uint8_t) constrain(lxWhite, 0, 255);; // *Added option to set White here, removed = 0
    return true;
  }
  return false;
}

void parseLxJson(int lxValue, byte segId, bool secondary)
{
  if (secondary) {
    DEBUG_PRINT(F("LY: Lox secondary = "));
  } else {
    DEBUG_PRINT(F("LX: Lox primary = "));
  }
  DEBUG_PRINTLN(lxValue);
  byte rgbw[] = {0,0,0,0};
  if (parseLx(lxValue, rgbw)) {
    if (bri == 0) {
      DEBUG_PRINTLN(F("LX: turn on"));
      toggleOnOff();
    }
    bri = 255;
    nightlightActive = false; //always disable nightlight when toggling
    DEBUG_PRINT(F("LX: segment "));
    DEBUG_PRINTLN(segId);
    strip.getSegment(segId).setColor(secondary, RGBW32(rgbw[0], rgbw[1], rgbw[2], rgbw[3])); // legacy values handled as well in json.cpp by stateUpdated()
  }
}

#endif
`

So I would like to have anyone else test this / I would like to see that feature being integrated soon. I tested cold / warm white, lower brightness etc.

FreakyJ, 2024-03-12:
Can't you use the option "Auto calculate white channel from RGB"?
[image]

blazoncek, 2024-03-12:
... and if you want brightest output just select Brighter.