wled/WLED #3768: Why ESP32-S3 does not supports analog audio input?

Suxsem, 2024-02-20:
### What happened?

I was reading https://github.com/Aircoookie/WLED/tree/main/usermods/audioreactive and noticed that ESP32-S3 does not supports analog audio input, only I2S.
Is this an hardware limitation or it has not been implemented yet?
Thank you

### To Reproduce Bug

-

### Expected Behavior

-

### Install Method

Binary from WLED.me

### What version of WLED?

-

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-20:
Analog is deprecated and will be removed in a future release. Internally I2S is used to process analog and S3 lacks that interface.

blazoncek, 2024-02-20:
Yes, it has I2S but IDF4.4.3 used for WLED lacks analog API AFAIK.

closed 2024-02-20 by blazoncek

softhack007, 2024-02-21:
> @blazoncek ESP32-S3 has I2S, I think new API should be used: [espressif/esp-idf#8536](https://github.com/espressif/esp-idf/issues/8536)

Its explained pretty well in the problem report you linked. We are using the I2S driver for audio input. On classic esp32 the I2S driver has a mode to read samples from ADC (analog input). This mode is not availeable on the new variants -S3/-S2/-C3. As a consequence, WLED audioreactive cannot support analog audio on these boards. 

The "continuous ADC" driver works completely different from I2S, so it's not an option for audioreactive.

Writing a new driver just for supporting ADC input would mean a lot of effort (complete re-write), just for supporting low quality input (ADC is very noisy, and only delivers 12bit instead of 24bit from I2S). We have decided that the effort needed for this is not justified; you can get I2S microphones and I2S line-in for cheap and with much better quality than any analog devices. 

Additionally, espressif has declared the "I2S-ADC" mode as depricted. It means that they are also not interested to support analog input in the future - if you don't like it, feel free to complain to espressif. If we would use ESP-IDF 5.x (latest version), then analog input on classic esp32 would be lost, too.