wled/WLED #3770: No mDNS broadcasts are sent from an ESP32 board when enabled

samdoran, 2024-02-22:
### What happened?

Configuring mDNS on an ESP32 board does not actually enable mDNS. It does not show up in [Discovery](https://itunes.apple.com/us/app/discovery-dns-sd-browser/id1381004916?mt=12) and a packet capture from the router interface shows no service discovery broadcasts from the ESP32 board. I see service broadcasts from ESP8266 devices running the same version of WLED.

### To Reproduce Bug

1. Navigate to Config > Wifi Setup
1. Enter an mDNS address
1. Click Save & Connect

### Expected Behavior

I am able to access the WLED device using its mDNS name.
A packet capture on the router interface shows traffic from the WLED device to `224.0.0.251:5353`.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
None but I'm happy to provide it if needed. I just need to get setup to capture the serial output.
```

### Anything else?

This seems exactly like #2944 which was resolved by changing to another ESP32 board.

I tested with 0.14.1, 0.14.2-b1, and 0.13.3 and the behavior is the same: no mDNS traffic is sent from the device.

This is happening on a [QuinLED-ESP32 board](https://quinled.info/quinled-esp32/). It's the only ESP32 board I have.

I have several ESP8266 boards running WLED 0.14.1 that work as expected.

<details>
  <summary>Packet capture of ESP8266 device</summary>
  
```
$ sudo tcpdump -ni bond0.111 port 5353 and host 10.11.2.196
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on bond0.111, link-type EN10MB (Ethernet), capture size 262144 bytes
21:26:06.369657 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
21:26:06.439277 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
21:26:06.442360 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
21:26:06.641608 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
21:26:06.691172 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
21:26:06.695991 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
21:26:06.892817 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
21:26:06.942560 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
21:26:06.945201 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
21:26:08.143894 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
21:26:08.194599 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
21:26:08.195825 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _http._tcp.local., PTR wled-control._http._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT (179)
21:26:09.174270 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
21:26:09.201149 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
21:26:09.201160 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _http._tcp.local., PTR wled-control._http._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT (179)
21:26:10.175448 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
21:26:10.201367 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
21:26:10.202149 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _http._tcp.local., PTR wled-control._http._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT (179)
21:26:11.174870 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
21:26:11.207979 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
[... 26 lines cut ...]
21:26:45.022132 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 3/0/4 PTR wled-control._wled._tcp.local., (Cache flush) TXT "mac=c45bbe55a971", PTR wled-control._http._tcp.local. (245)
21:26:45.033907 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 3/0/4 PTR wled-control._wled._tcp.local., (Cache flush) TXT "mac=c45bbe55a971", PTR wled-control._http._tcp.local. (245)
```
</details>

<details>
  <summary>Packet capture of ESP32 device</summary>

```
$ sudo tcpdump -ni bond0.111 port 5353 and host 10.11.2.215
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on bond0.111, link-type EN10MB (Ethernet), capture size 262144 bytes
# Nothing :(
```
</details>

Configuration:
[image]

All DNS-SD services on my network.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-02-22:
@samdoran Might be your router is to blame as it works on my setups
[image]

samdoran, 2024-02-22:
It could be something in my environment. I just find it odd that ESP8266 devices on the same VLAN work as expected but the ESP32 device does not. I just don't see any traffic coming from the ESP32 device like I do the ESP8266 devices.

@dosipod Is that an ESP32 device?

dosipod, 2024-02-22:
@samdoran Yes all the  devices i am using  are ESP32 as i no longer use ESP8266 with wled  . Also my router is a cheap TL-WR840N ( I used it  after facing few issues with my iot devices on Apple time capsule using ddp , sync ..etc  ) . If  you are having lot of devices might be it is worth it to use a separate less restrictive router

samdoran, 2024-02-22:
I wonder if it's something specifically with the QuidLED ESP32.

From the serial console, I can see that mDNS is starting, but it just doesn't send any mDNS traffic. Quite bizarre.

```
Connected! IP address: 10.11.2.215
Init STA interfaces
mDNS started
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
```

When I reboot an ESP8266 device, the first traffic I see is mDNS

<details>
  <summary>ESP8266 after boot</summary>

```
12:31:39.032031 IP 10.11.0.136.49394 > 10.11.2.196.80: Flags [S], seq 269630730, win 29200, options [mss 1460,sackOK,TS val 1212080780 ecr 0,nop,wscale 7], length 0
12:31:39.806775 IP 10.11.2.129.67 > 10.11.2.196.68: BOOTP/DHCP, Reply, length 300
12:31:39.820906 IP 10.11.2.129.67 > 10.11.2.196.68: BOOTP/DHCP, Reply, length 300
12:31:39.826485 ARP, Request who-has 10.11.2.196 tell 10.11.2.196, length 46
12:31:39.839306 IP 10.11.2.196.65506 > 255.255.255.255.65506: UDP, length 44
12:31:40.017519 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
12:31:40.088664 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
12:31:40.092136 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
12:31:40.272071 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
12:31:40.341900 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
12:31:40.345356 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
12:31:40.519877 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [1n] ANY (QU)? wled-control.local. (70)
12:31:40.602858 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._wled._tcp.local. (189)
12:31:40.605680 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0 [2n] [2au] ANY (QU)? wled-control._http._tcp.local. (172)
12:31:41.771462 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
12:31:41.854796 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
12:31:41.857280 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _http._tcp.local., PTR wled-control._http._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT (179)
12:31:42.773237 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 2/0/0 (Cache flush) A 10.11.2.196, (Cache flush) PTR wled-control.local. (84)
12:31:42.856315 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _wled._tcp.local., PTR wled-control._wled._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT "mac=c45bbe55a971" (196)
12:31:42.857111 IP 10.11.2.196.5353 > 224.0.0.251.5353: 0*- [0q] 4/0/1 PTR _http._tcp.local., PTR wled-control._http._tcp.local., (Cache flush) SRV wled-control.local.:80 0 0, (Cache flush) TXT (179)
12:31:43.063985 IP 10.11.0.136.49394 > 10.11.2.196.80: Flags [S], seq 269630730, win 29200, options [mss 1460,sackOK,TS val 1212084812 ecr 0,nop,wscale 7], length 0
12:31:43.069884 IP 10.11.2.196.80 > 10.11.0.136.49394: Flags [S.], seq 6510, ack 269630731, win 2144, options [mss 536,nop,nop,sackOK], length 0
```
</details>

But the ESP32 board doesn't send any mDNS traffic even though the serial console states that the process started.

<details>
  <summary>ESP32 packet capture after boot</summary>

```
12:36:30.987354 IP 10.11.2.129.67 > 10.11.2.215.68: BOOTP/DHCP, Reply, length 315
12:36:31.006469 IP 10.11.2.129.67 > 10.11.2.215.68: BOOTP/DHCP, Reply, length 315
12:36:31.011218 ARP, Request who-has 10.11.2.215 tell 10.11.2.215, length 46
12:36:31.041201 IP 10.11.2.215.65506 > 255.255.255.255.65506: UDP, length 44
12:36:35.755925 IP 10.11.1.55.49833 > 10.11.2.215.80: Flags [S], seq 2305434687, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 275613416 ecr 0,sackOK,eol], length 0
12:36:35.756110 IP 10.11.1.55.49834 > 10.11.2.215.80: Flags [S], seq 3620895774, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 3997485371 ecr 0,sackOK,eol], length 0
12:36:35.756303 IP 10.11.1.55.49835 > 10.11.2.215.80: Flags [S], seq 2125630009, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 195368913 ecr 0,sackOK,eol], length 0
12:36:35.756526 IP 10.11.1.55.49836 > 10.11.2.215.80: Flags [S], seq 1989167518, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 4064554006 ecr 0,sackOK,eol], length 0
12:36:35.773850 IP 10.11.2.215.80 > 10.11.1.55.49833: Flags [S.], seq 4148016308, ack 2305434688, win 5744, options [mss 1436], length 0
12:36:35.774506 IP 10.11.1.55.49833 > 10.11.2.215.80: Flags [.], ack 1, win 65535, length 0
12:36:35.779419 IP 10.11.2.215.80 > 10.11.1.55.49834: Flags [S.], seq 1935070257, ack 3620895775, win 5744, options [mss 1436], length 0
12:36:35.780193 IP 10.11.2.215.80 > 10.11.1.55.49835: Flags [S.], seq 1463800227, ack 2125630010, win 5744, options [mss 1436], length 0
12:36:35.780352 IP 10.11.1.55.49834 > 10.11.2.215.80: Flags [.], ack 1, win 65535, length 0
12:36:35.780742 IP 10.11.1.55.49835 > 10.11.2.215.80: Flags [.], ack 1, win 65535, length 0
12:36:35.803265 IP 10.11.2.215.80 > 10.11.1.55.49833: Flags [.], seq 1:1437, ack 398, win 5347, length 1436: HTTP: HTTP/1.1 200 OK
[... cut ...]
12:36:35.864397 IP 10.11.2.215.80 > 10.11.1.55.49835: Flags [.], seq 1:1437, ack 316, win 5429, length 1436: HTTP: HTTP/1.1 200 OK
12:36:35.864603 IP 10.11.2.215.80 > 10.11.1.55.49835: Flags [P.], seq 1437:2124, ack 316, win 5429, length 687: HTTP
12:36:35.867219 IP 10.11.1.55.49835 > 10.11.2.215.80: Flags [F.], seq 316, ack 2124, win 65535, length 0
12:36:35.880722 IP 10.11.2.215.80 > 10.11.1.55.49835: Flags [F.], seq 2124, ack 317, win 5428, length 0
```
</details>

Are there any more detailed logs available to see if maybe the mDNS process is failing to start for some reason?

samdoran, 2024-02-23:
I've done a lot more analysis on this. When the ESP32 device and a laptop are connected to a GL-iNet router, things work as expected.

My main network has APs connected via trunk ports to a switch, and the switch is connected via trunk ports to the router.

Even with a laptop connected directly to the switch in the same layer two space as the EPS32 device, I only see the mDNS traffic on the laptop _after_ unplugging/plugging the interface. Quite bizarre.

This seems to indicate the switch is handling the mDNS traffic from the ESP32 device differently than the traffic from the ESP8266 devices. It's a managed switch (Cisco 3750-X), so it's possible there is a configuration change I need to make. I tried adjusting IGMP settings but that did not make a difference.

I compared packet captures of mDNS traffic from the ESP8266 device and the ESP32 device. They are different. I don't fully understand the differences at this point.

This does not seem to be a WLED bug but is most likely an issue or intentional difference in the library WLED is using for mDNS.

If I ever figure out how exactly to make this work, I'll report back. Thanks for your help.

closed 2024-02-23 by samdoran