wled/WLED #3771: AutoSave mod doesn't autosave

markg85, 2024-02-22:
### What happened?

Installed the AutoSave usermod with these flags:
```
-D USERMOD_AUTO_SAVE -D AUTOSAVE_AFTER_SEC=10 -D AUTOSAVE_PRESET_NUM=100 -D USERMOD_AUTO_SAVE_ON_BOOT=true
```

Relevant screenshots:
(presets section on the main page)
[image]

(Relevant section from "LED Preferences")
[image]

(usermod)
[image]

Am i missing a setting somewhere?

### To Reproduce Bug

Install the usermod with these flags:
```
-D USERMOD_AUTO_SAVE -D AUTOSAVE_AFTER_SEC=10 -D AUTOSAVE_PRESET_NUM=100 -D USERMOD_AUTO_SAVE_ON_BOOT=true
```

Create a preset with id 100 and some effect
Change your effect and wait for at least `AUTOSAVE_AFTER_SEC=10` (10 seconds).
Reboot.

You now get your initial effect in id 100 where i was expecting autosave kick in and **save** the change i made automatically.

### Expected Behavior

Preset loads at boot.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.2-b1 (build 2402120)

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Anything else?

I did compile from source so this is the current git version as of the moment of this post.

AutoSave is freakingly annoying to setup. It's completely unclear to me if i already need to have a preset to which i "tie" autosave. If i don't have a preset then i get a weirdly named one created by autosave. I did try that path and that only saved my initial settings. It didn't update - autosave - any changes i made.

In the compile flags i explicitly said `-D AUTOSAVE_PRESET_NUM=100` which is ignored in the usermod gui. It's default is 250. Therefore i'm not at all sure if what i set in the compile flags is even properly used in autosave. Clearly the flags are in the right place else the whole usermod wouldn't be visible (compiled in).

My assumption is AutoSave to update my preset with new changes. It doesn't need to do anything else as the preset to load at boot is already set in the LED Preferences. If this assumption is correct, how do i get AutoSave to play well?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-02-22:
Hi, 

Expected behaviour of autosave is described here:
https://github.com/Aircoookie/WLED/tree/main/usermods/usermod_v2_auto_save#auto-save

In fact the usermod is not meant to auto-update your presets. However it can be used to regularly write your current settings into a special preset number called "autosave preset" (default = 250). The preset name will reflect the time&date of last autosave (format: `~ MM-DD HH:MM:SS ~`)

There are a few limitations however

https://github.com/Aircoookie/WLED/blob/c8c394b4e923d98d37b95e6007509b40b624a8a3/usermods/usermod_v2_auto_save/usermod_v2_auto_save.h#L125

* if you're running a preset without changes, then autosave will not save the settings (reason: you're running an unchanged preset and we don't want to make duplicates)
* sometimes the delay until auto-save is longer, especially if you have a long LED stripe (> 400 leds approx) or running at very high fps.

markg85, 2024-02-23:
Hi @softhack007,

Thank you for that reply!

Nothing you described there is clear from the documentation you mention (what you said there was 1000x more helpful then said documentation). The current documentation is vague at best and open to very broad guessing to even get it to work.

That 250 default value - when specified 100 as compile define - must be a bug. I didn't change any GUI values yet the gui said 250.

With that being said, it turns out the settings are auto saving. It just takes many minutes (haven't figured out how many) but at least more than a couple and less then 15.

markg85, 2024-02-23:
Hmm, no, it doesn't work.

I just want the preset to change to whatever i click on. That can't be that difficult?!?!

Stuff like `strip.isUpdating()` should - from just reading code - be completely irrelevant. Unless the current autosave tries to save the state of each individual pixel. But that isn't what i want nor need.

Imho, this should be a core feature, not a usermod.

blazoncek, 2024-02-23:
> Edit: @blazoncek this is another usermod that has `if (strip.isUpdating()) return;` without a time limit. We might need to change that, to have a time limit for "be nice but not too nice" (similar to multirelay)

Or remove it altogether.

blazoncek, 2024-02-23:
@markg85 usermods are extensions written by WLED users and core team is not responsible for their content or description or functionality. We do try, however, to keep them working when breaking changes occur but only if we (ourselves) find them useful and worth keeping.

You should direct your questions, complaints and remarks to original author of the usermod.

closed 2024-06-29 by github-actions[bot]