wled/WLED #3772: WLED random reboots when MQTT is enabled

hajeth, 2024-02-24:
### What happened?

When MQTT is enabled WLED crashes after some minutes and reboots. 
Eventually identical to Issue #3435

### To Reproduce Bug

Set the MQTT Broker and activate the MQTT feature - then reboot to enable MQTT and wait

### Expected Behavior

No crash/reboot

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.2-b1 2402120

### Which microcontroller/board are you seeing the problem on?

ESP32, Other

### Relevant log/trace output

```shell
WLED log with AsyncTCP logging enabled:

event arg == NULL: 0x3ffdfcac
event arg == NULL: 0x3ffdfcac
event arg == NULL: 0x3ffdfcac
Reconnecting MQTT
C: 0x3ffdcbe4 0x3ffdcec8 0
+F: 0x3ffdfcac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4012088b  PS      : 0x00060230  A0      : 0x80120a43  A1      : 0x3ffb5ce0
A2      : 0x00000000  A3      : 0x3ffdfcac  A4      : 0x00000000  A5      : 0x00000063  
A6      : 0x00000000  A7      : 0x3ffb5c49  A8      : 0x00000000  A9      : 0x3ffb5c50
A10     : 0x0000000a  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00001800  
A14     : 0x3ffb5c47  A15     : 0x3ffb5280  SAR     : 0x00000005  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40120888:0x3ffb5ce0 0x40120a40:0x3ffb5d00 0x40120a9e:0x3ffb5d20 0x40156fe9:0x3ffb5d50 0x4015d26f:0x3ffb5d70 0x40161c1a:0x3ffb5da0 0x4014be2b:0x3ffb5dc0 0x4008b936:0x3ffb5df0
  #0  0x40120888:0x3ffb5ce0 in AsyncClient::_lwip_fin(tcp_pcb*, signed char) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1153
  #1  0x40120a40:0x3ffb5d00 in AsyncClient::_s_lwip_fin(void*, tcp_pcb*, signed char) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1153
  #2  0x40120a9e:0x3ffb5d20 in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1153
  #3  0x40156fe9:0x3ffb5d50 in tcp_input at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:475 (discriminator 1)
  #4  0x4015d26f:0x3ffb5d70 in ip4_input at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:750
  #5  0x40161c1a:0x3ffb5da0 in ethernet_input at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:184
  #6  0x4014be2b:0x3ffb5dc0 in tcpip_thread at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
  #7  0x4008b936:0x3ffb5df0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
```

### Anything else?

ESP Type: ESP32-D0WDQ5 rev.3
Mosquitto 2.018, running on Windows Server 2019 also tried Bevywise Broker with same result

I am aware that the exception occurs in AsyncTCP and will also try to contact the programmer regarding this issue.

But every time the reboot occurs there is a "Reconnect MQTT" Message before the crash and i am wondering why WLED issues MQTT reconnects, when already connected. Also it seems it does 2 connection attempts (The Mosquitto log shows incoming connections from two TCP Ports to Port 1883

Mosquitto Log:

> 1708761740: New connection from 192.168.17.28:56454 on port 1883.
1708761740: New connection from 192.168.17.28:56455 on port 1883.
1708761740: Client WZ_WLED_TV already connected, closing old connection.
1708761740: New client connected from 192.168.17.28:56454 as WZ_WLED_TV (p2, c1, k60).
1708761740: Will message specified (7 bytes) (r1, q0).
1708761740: wled/WZ_WLED_TV/status
1708761740: Sending CONNACK to WZ_WLED_TV (0, 0)

The packet trace shows a successful MQTT connection (Packet No 75), then the ESP32 is not available anymore due the reboot

[image]

Logs and packettrace attached

Many thanks for any help to solve this tricky one
[Attachments.zip](https://github.com/Aircoookie/WLED/files/14392291/Attachments.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-24:
Closing as a duplicate of #3435
Not part of WLED code.

closed 2024-02-24 by blazoncek

hajeth, 2024-02-24:
The question was, why WLED reconnects to the mqtt broker even when it is already connected and why there are two connections on two ports??