wled/WLED #3773: Rearranging presets

B1BU, 2024-02-24:
**Is your feature request related to a problem? Please describe.**
Rearranging presets requires juggling IDs and is very tiresome.

**Describe the solution you'd like**
An option to rearrange presets and have WLED automatically update their IDs to reflect the changes.
Changes could be saved locally until you applied them to avoid issues.

**Additional context**
I use my WLED setup most of the day since it's my preferred light source in my room. I control it with an IR remote most of the time and because of that it's important that my presets are well organized for ease of access when cycling between them.

Minimemat, 2024-02-27:
I'm confused, what are you asking for? Presets are already able to be sorted by ID

Sorting the effect list by ID would be great too

B1BU, 2024-02-27:
Being able to move presets up and down in the list and/or have their ids change accordingly

Minimemat, 2024-02-27:
I'm sure someone can come up with an easier solution but if you modify the id and then hit SAVE it will copy it to that new id. You can then delete the old one

B1BU, 2024-02-27:
I coded a script for that (that didn't work until I figured out what was causing it #3780), but I wish there was a built-in way to do it

blazoncek, 2024-02-27:
Not likely to happen any time soon, unless you tackle it yourself and make a PR. Sorry.