wled/WLED #3778: "Restore configuration" accepts preset backup files, bricking the WLED device

B1BU, 2024-02-26:
### What happened?

I accidentally uploaded a _presets.json_ file as a _cfg.json_ file in the Security & Updates section of the settings menu and that bricked my WLED device. The two options look very similar and are very close together so this is a very easy mistake to make.

I'll fix it when I get home, but I just wanted to let you guys know that this is an issue.

### To Reproduce Bug

Upload a _presets.json_ or any .json file in the "Restore configuration" option of the Security & Updates section of the settings menu.

### Expected Behavior

One or more of the following:
1. For the device to not accept the file.
2. For the control panel to show a warning that the file is invalid.
3. For the device to save a temprary backup of the settings before replacing the file and restore it if the uploaded file doesn't work, otherwise deleting the backup to free up storage.
4. For the two options in the settings menu to be further apart and/or look visually distinct.
5. For the warning text to be at the top of the Backup & Restore section maybe?

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.2-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-26:
That is the expected behaviour. Unfortunately.
There is a warning saying so.
[image]

B1BU, 2024-02-26:
Yeah, but my issue is that it's still too easy to make the mistake. The options could at least be further apart or look more visually distinct.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

commit 0ff5dec0 (2024-02-28): Reposition upload warning message.
Fixes #3778
closed 2024-03-22 by blazoncek