wled/WLED #3780: Presets file gets corrupted when presets are modified if it contains any spacing

B1BU, 2024-02-27:
### What happened?

If a preset file contains white spaces it loads up properly but gets corrupted when any preset is edited.

Writing the file like [`{"key":"value"}`](https://github.com/Aircoookie/WLED/files/14414275/works.json) works.

But writing it like [`{"key": "value"}`](https://github.com/Aircoookie/WLED/files/14414274/broken.json) doesn't.

It seems WLED tries to overwrite the length of the preset data with spaces and add the new data to the end of the file but it doesn't account for white spaces, so it overwrites the wrong part of the file creating invalid json syntax.

I don't know why you guys chose to overwrite the file like this and I imagine there's probably a reason, but I really wish it would just write the file properly like it does with config files. People have always complained preset files would get corrupted randomly, this seems to be the reason why.

### To Reproduce Bug

Add a single space between any key and value of any preset in presets.json and try to modify the same preset.

### Expected Behavior

For WLED to the file properly like it does with cfg.json or at the very least support spacing.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.2-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

People have always complained that presets in WLED can get corrupted randomly. This seems to be why

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-02-27:
Unfortunately, WLED requires minimized JSON files (especially presets.json as it modifies it).
Modifying this behaviour will most likely never be fixed as it is very convoluted to do otherwise without affecting performance.

B1BU, 2024-02-27:
could you explain why it replaces the preset data with spaces? that seems like it just makes the file bigger for no reason

blazoncek, 2024-02-27:
Because of speed/efficiency.

closed 2024-07-04 by github-actions[bot]