wled/WLED #3785: Some effects on the wiki are outdated

LightAndy1, 2024-02-29:
### What happened?

I searched for the 49th effect (Fairy) on the wiki and it doesn't match. The 49th effect on the wiki is the Police effect.

### To Reproduce Bug

Self explanatory

### Expected Behavior

The wiki should be updated as it's a bit out of date.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-01:
@scottrbailey can you take a look when time permits?

@LightAndy1 next time please open the issue in correct repository. In this case WLED-Docs.

closed 2024-03-01 by blazoncek

scottrbailey, 2024-03-01:
The wiki that is deprecated and read only?

Actually, why is it still around?

blazoncek, 2024-03-01:
Oops! My bad.
Yes, wiki is obsolete..

LightAndy1, 2024-03-01:
Oh, well sorry guys :(