wled/WLED #3787: Cycle through Presets via Poti / Button

robertvb83, 2024-03-02:
Hi,

title says it... It would be cool to have a nice Dial aka Poti to cycle through your list of presets.

Divide Voltage range by number of presets and use each slot for one preset.

In addition I would like a Button not to be assigned to one specific preset but go to the next preset on every press.

Regards
Robert

blazoncek, 2024-03-02:
Please see [KB](https://kno.wled.ge).
And/or use WLED [forum](https://wled.discourse.group) for help and support questions.

closed 2024-03-02 by blazoncek

robertvb83, 2024-05-04:
The same question was asked here https://github.com/Aircoookie/WLED/issues/3932

Only for reference:

I found what i was looking for:

That exact functionality is actually already available and is described here in the presets section
https://kno.wled.ge/features/presets/

Scroll down to the last entry of the table. You only need to define some presets and then create a special cycling preset. This preset is then assigned to your button.

[image]