wled/WLED #3792: BTNPIN flag does not work for multiple buttons

ZOUDIO, 2024-03-04:
### What happened?

The BTNPIN flag does not seem to be working for multiple pins. Only the first pin gets added.

### To Reproduce Bug

I used the buildflag '-D BTNPIN=14,32' but only 14 shows up in the GUI
[image]

### Expected Behavior

I would expect multiple buttons to be initialized and visible in the web GUI.

### Install Method

Self-Compiled

### What version of WLED?

v0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
-
```

### Anything else?

I used the same type of buildflag for multiple LED output pins, and this is working:
-D DATA_PINS=2,4,12,13,15,17

So it is specifically related to this build flag it seems

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek

Arcitec, 2025-04-02:
@ZOUDIO Your example is missing `BTNTYPE`. You must set that too, if you want to pre-compile multiple pins. Otherwise the other button pins will default to "disabled".

Check the source code for values, but the most common value is `2` (Pushbutton), so in your case you would have to do:

```ini
  -D WLED_MAX_BUTTONS=2
  -D BTNPIN=14,32
  -D BTNTYPE=2,2
```