wled/WLED #3793: Audioreactive does not work on IO34

ZOUDIO, 2024-03-05:
### What happened?

ADC reports no activity on IO34 when using the Audioreactive usermod. Empty sketch which prints the analogRead result says the analog signal is just fine. WLED reads the signal just fine if I move the signal to IO33.

### To Reproduce Bug

Run audioreactive with the analog input on IO34 and viewing the info page of WLED will report 'ADC quiet'. Moving the same signal to IO33 makes it work.

### Expected Behavior

IO34 is an ADC pin and should work

### Install Method

Self-Compiled

### What version of WLED?

v0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

IO34 is an ADC pin and should work

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-03-05:
Hi, 

I assume you want to use analog microphone, correct? which microphone exactly?
Actually the audioreactive source code handles all ADC1 pins the same way - so its still a mistery to me why one specific pin works while another does not work.  It could be that your board has a special purpose assigned to  gpio34 - like battery voltage or differential ADC. 

* please inform us about the kind of microphone board you use. We recommend MAX9814, anything else means trouble.
* Please check if audioreactive works with pin 32 or pin 36. This might help to better understand if your board has some special circuitry on gpio34.

There is some help on connecting analog inputs here: https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options

ZOUDIO, 2024-03-18:
I did some more experimentation and I also got it to work on IO34.
It's quite difficult to replicate but I am quite sure something goes wrong when changing the pins using the GUI. 
Sometimes the audioreactive enable checkbox is not enabled on the settings page, and when I change the pin number it does not work anymore, even after reboot or power cycle.
Directly after uploading new firmware it always works, regardless of the pin.
I am afraid I do not have more time to look into this issue, maybe you are able to replicate it, but otherwise this can be closed as the issue for my usecase is already solved.

closed 2024-03-18 by blazoncek