wled/WLED #3796: WLED CLOCK WITH RTC USERMOD

Jagannathnkulkarni, 2024-03-06:
Kindly provide complied binary to use with WLED CLOCK WITH RTC SUPPORT usermod as it is nearly impossible to compile the code myself by any of the method described here. It will be a great favour for many of us. Thanks !!

blazoncek, 2024-03-06:
Unfortunately, no. We will not provide such binary as then we'd need to provide thousands of different binaries for each user requesting a special combination of usermods and/or features.

Please use Discord or WLED [forum](https://wled.discourse.group) for help and support questions.
I'm sure there will be someone kind enough to provide you with what you need.

Jagannathnkulkarni, 2024-03-06:
Thanks for your reply buddy.

It's perfectly ok. I can understand.

and still many thanks for your awesome work. God bless you. !!