wled/WLED #3803: ArduinoFFT 2.0.0 release breaks audioreactive (even though build system should prevent it)

Xevel, 2024-03-09:
### What happened?

3 days ago, kosme/ArdunioFFT released version 2.0.0. This is a breaking change in the API, but that is not the topic of this bug report.

The problem is that the line declaring the dependency in platformio.ini now causes (on a fresh install) the following error:

`AR_lib_deps = https://github.com/kosme/arduinoFFT#develop @ ^1.9.2`

> PackageException: Package version 2.0.0+sha.909240f doesn't satisfy requirements ^1.9.2 based on PackageMetadata <type=library name=arduinoFFT version=2.0.0+sha.909240f spec={'owner': None, 
> 'id': None, 'name': 'arduinoFFT', 'requirements': '^1.9.2', 'uri': 'git+https://github.com/kosme/arduinoFFT#develop'}

Looking at releases on kosme/ArdunioFFT, I do not understand why it used to work: there are no releases or tags between 1.6.2 and 2.0.0, be it on master or develop branches, so how did platformIO not complain when we previously asked for 1.9.2 or newer ?
It did indeed find "arduinoFFT @ 1.9.2+sha.419d7b0" according to my other, older copy of wled repo where the .pio folder has not been fully cleaned...

How to get the 1.9.2 version that does not exist according to kosme/arduinoFFT back until(if) audioreactive is ported to 2.0.0?

**What I have tried to fix:**
I tried removing the ^, thus theoretically forcing the specific version instead of "this version or newer that does not break API" but platformio still downloaded 2.0.0 then complained.

** Misc questions **
And for my own curiosity: why on a new install it now cannot find it ? why does it download the 2.0.0 even if according to the [platformio version requirement documentation](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#cmd-pkg-install-requirements) it should not ?

### To Reproduce Bug

Either start from a fresh clone of wled repo, or on an existing local copy, do a Full Clean.
Build for target esp32dev_audioreactive

### Expected Behavior

Release of newer (API breaking) ArduinoFFT versions should not break existing build system.

### Install Method

Self-Compiled

### What version of WLED?

v0.14.2-b1  d9b0df94e2a344b05bac88270a162b8e25f86ec5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
PackageException: Package version 2.0.0+sha.909240f doesn't satisfy requirements ^1.9.2 based on PackageMetadata <type=library name=arduinoFFT version=2.0.0+sha.909240f spec={'owner': None, 
id': None, 'name': 'arduinoFFT', 'requirements': '^1.9.2', 'uri': 'git+https://github.com/kosme/arduinoFFT#develop'}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Xevel, 2024-03-09:
Ok I found the fix:

`AR_lib_deps = https://github.com/kosme/arduinoFFT#419d7b0`

This will download and use the elusive 1.9.2 version (naming it by its commit instead of tag or release version...), same as before, and allow compilation to proceed normally.

blazoncek, 2024-03-09:
0.15 has that fixed. ATM We'll not be fixing for prior release.

FYI ^ means any greater version.

Xevel, 2024-03-09:
Makes sense.

From the platformio version requirement documentation linked above:

> ^1.2.3 - any compatible version (new functionality in a backwards compatible manner and patches are allowed, 1.x.x).

You're confusing ^ with > : 

> \>1.2.3 - any version greater than 1.2.3. >=, <, and <= are also possible

closed 2024-03-09 by Xevel