wled/WLED #3806: Feature request: Easier way to set preset at boot

blitzmann, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
Other than bad UX, no

**Describe the solution you'd like**
I am learning about WLED, and as I go through it I noticed tat every time the board lost power, the LED settings that I had revert to the default (a solid orange color). It took a while to track down how to get this to boot to my preferences - I read issues from the past that specified that you needed to compile your own version that has a user save mod, etc. Turns out you need to do the following:

1) Save a preset
2) Note the ID of the preset
3) Go to Settings > LED Preferences
4) Apply preset at boot

As a newer user, I assumed it would save whatever I had active - it never occurred to me to create a preset. I think there can be a few different things that could be implemented to improve the UX with this:

* Be able to set a preset to the "boot" preset right from the preset section.
    * Avoids having to go dig for it in the options. 
* A notice or banner at the top, explaining that changes will not persist through a powercycle
    * Could also link to a wiki page that explains this

**Additional context**
[image]

I would be interested in possibly tackling something like this. I would need some guidance on how best to implement this. My specialty lies in web frameworks and C# backends. Not sure how the web pages are constructed here.

blazoncek, 2024-03-09:
That's all written in [KB](https://kno.wled.ge).

blitzmann, 2024-03-09:
I'm sure it's there, but only if you know about the knowledge base, and only if you know where to look (probably in the "preset" page, but you would need to know that's what you're looking for). You still have to really dig for it, instead of it being an intuitive interface. I would be willing to put in some time to help with this if there is interest.

blazoncek, 2024-03-10:
You can help by either changing the code or updating the documntation to make it more obvious.

blazoncek, 2024-03-10:
[image]

blitzmann, 2024-03-10:
Yeah, that's pretty much what I envision! An icon on the header to denote it as the boot preset would also go a long way in case you have many presets. Is this a commit on a branch somewhere, or just a mockup?

I started digging into the frontend code as well. It's a doozy, I don't remember the last time I didn't work with a framework, but seems pretty straightforward in finding where things live.

blitzmann, 2024-03-10:
I would also ask if there is a way to serve this in a mock fashion, instead of building it and pushing it to a esp to test. I'll look around the docs, but i didn't see anything regarding that in CONTRIBUTING docs

willmmiles, 2024-03-10:
> I would also ask if there is a way to serve this in a mock fashion, instead of building it and pushing it to a esp to test. I'll look around the docs, but i didn't see anything regarding that in CONTRIBUTING docs

If you open wled00/data/index.htm in a browser right off your working copy, it prompts you for an IP and then uses the device data from there.  (The config panels redirect to the actual device.)

blitzmann, 2024-03-11:
Thanks for the advise, works great!

So it seems that most things are global variables. When config changes, a `doSerializeConfig` flag is set which is handled in the next loop, and saves all the global setting variables to a cfg.json file.

This is definitely a pattern I'm not used to, but I think all I would need to do is set a flag to the the preset save request, intercept that on the controller, and then set the `bootPreset` and `doSerializeConfig` flags so that it's saved. I'm still hunting down where that preset save is handled - looks like it's a web socket request, but trying to find out where it's actually handled.

There also seem to be a lot of magic strings that are just a few characters used for keys, which makes it hard to do a global search to dig deeper. Not sure if this is for specific reasons (space / memory) or not.

blazoncek, 2024-03-11:
It is not a mockup. I have working version locally, waiting to be pushed.
You got the rough workings all right.

blazoncek, 2024-03-11:
[image]

Doyle4, 2024-03-12:
> ![Screenshot 2024-03-10 at 20 06 08](https://private-user-images.githubusercontent.com/7722392/311541233-1a2f21b7-e333-4c4c-b90e-fe336f827a29.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyMDk5NTgsIm5iZiI6MTcxMDIwOTY1OCwicGF0aCI6Ii83NzIyMzkyLzMxMTU0MTIzMy0xYTJmMjFiNy1lMzMzLTRjNGMtYjkwZS1mZTMzNmY4MjdhMjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTJUMDIxNDE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2Y3Y2EwM2M4ZWI2MTA5ZWM2ODBiZDg0ODkxODJhMWQwYzE2NDI2ZDVhNmU1NjliYTlkYWIyMzUyZDE1YTdhZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hF3F60CJH7v5WbWxxxYDT3OZkrph_DKLvN0jfxUMDQE)

Now this is something Im very interested in also, I know where to go and find the boot option, but to not leave the main page would be amazing. Good request imo.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

commit a4384bd3 (2024-03-11): Simpler boot preset save
- fixes #3806
closed 2024-03-22 by blazoncek