wled/WLED #3807: Big bug Sound reactive stop working after 2 days

parham3423, 2024-03-09:
### What happened?

Sound reactive function stop working after 2 days 
And nothing make it work , changing mic pin , reset , disconnect from power source, ota manual update to flash firmware (also flashed binary from wled using ota)
all other function worked except sound reactive . 
The only fix was to erase flash , and upload it again with platformio
Version : Wled 0.14.1
Device : esp32 Dev

### To Reproduce Bug

Don't know what happened . It was working fine and  suddenly stop working , I take down all of thing and did what I told you 

### Expected Behavior

Do not stop working after some time

### Install Method

Self-Compiled

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-03-15:
Hi, a few questions to help me reproduce your issue:
* What kind of microphone are you using?
* Which PlatformIO environment did you use for compiling?
* Did you upload directly from PlatformIO, or did you install WLED by OTA?

blazoncek, 2024-06-26:
Closing as no response received.

closed 2024-06-26 by blazoncek