wled/WLED #3808: Value for Scrolling Text

LosFritzos, 2024-03-10:
Hi there,
Is it possible to implement a value for the text of the "scrolling text" effect?
Currently this is defined via the segment name, which unfortunately I cannot change using the ioBroker Wled adapter from Drozmotix.
It would be a really cool feature because I could display all the values ​​from my smart home in a very nice Way on the 2D Led Matrix
Best regards

Aircoookie, 2024-03-10:
Hi, what do you mean with "implement a value"? If we added a separate variable to the API to set the "Scrolling Text" text, this would have to be added to the ioBroker adapter as well, at which point they could (and should!) just add support for setting the segment name.
IMO segment name is quite an elegant and lightweight solution for the text and will also be used by other effects in similar ways, for example for the upcoming "Image" effect to determine which .gif file to display.

closed 2024-03-10 by Aircoookie

LosFritzos, 2024-03-10:
sorry, maybe I expressed myself wrong, that was the answer to my question to DrozmotiX whether it was possible to change the segment name via the ioBroker adapter: "So please post an enhancement directly to WLED to make it possible to change this value by the API and we can implement it to the adapter."

blazoncek, 2024-03-10:
[KB](https://kno.wled.ge/interfaces/json-api/)