wled/WLED #3809: Loxone JSON parser doesn't handle lx=0 correctly

FreakyJ, 2024-03-10:
### What happened?

I connected WLED to Loxone using the UDP Jason API to control individual segments:
`{"seg":[{"id":0,"lx":<v>}]}`
However, when I tunred the stripe off in loxone, it never turned off... I then debuged it using CURL:
`curl -X POST -H "Content-Type: application/json" -d '{"seg":[{"id":0,"lx":0}]}' 192.168.3.245/json`
I got success, but nothing changed.
Using this command, the stripe looked off (change: "lx":1):
`curl -X POST -H "Content-Type: application/json" -d '{"seg":[{"id":0,"lx":1}]}' 192.168.3.245/json`

I checked the code and in the json.cpp in line 191 and 196 I saw:
`if (lx > 0) {

if (ly = 0) {`
but it mus be greater or equal, since loxone sends a 0 to turn everything off.
Changing the code to this works fine with loxone:

`// lx parser
  #ifdef WLED_ENABLE_LOXONE
  int lx = elem[F("lx")] | -1;
  if (lx >= 0) {
    parseLxJson(lx, id, false);
  }
  int ly = elem[F("ly")] | -1;
  if (ly >= 0) {
    parseLxJson(ly, id, true);
  }
  #endif`

### To Reproduce Bug

send `curl -X POST -H "Content-Type: application/json" -d '{"seg":[{"id":0,"lx":0}]}' 192.168.3.245/json` stripe won't turn off

### Expected Behavior

stripe should turn off

### Install Method

Self-Compiled

### What version of WLED?

0.14.2-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-12:
I never used Loxone so bear with me.

There is no logic within WLED to turn WLED off from Loxone and there never was (explicitly).

```c++
  // lx parser
  #ifdef WLED_ENABLE_LOXONE
  int lx = elem[F("lx")] | -1;
  if (lx > 0) {
    parseLxJson(lx, id, false);
  }
  int ly = elem[F("ly")] | -1;
  if (ly > 0) {
    parseLxJson(ly, id, true);
  }
  #endif
```

So I do not know if this is a bug or not but the last change to Loxone code was 3 years ago.

blazoncek, 2024-03-12:
If you do think this will solve Loxone problem, open a PR, please.

FreakyJ, 2024-08-14:
I've created the pull request, sorry that it took so long...
https://github.com/Aircoookie/WLED/pull/4102

commit 198aceee (2024-08-16): Merge pull request #4104 from FreakyJ/Fix-Bug-#3809

Fixes #3809 Loxone JSON parser doesn't handle lx=0 correctly
closed 2024-10-27 by blazoncek