wled/WLED #3811: Multiple Segments with 1D and 2D arrangements

TheLastMuri, 2024-03-11:
**Is your feature request related to a problem? Please describe.**
I tried to set up a micture of a 1D Stripe at the beginning which leads into a 15x15 Matrix. Haven't found any possibility to have it set up.

**Describe the solution you'd like**
Having the possibility to enter a total amount of LEDs and divide it into:
Y LEDs = 1st segment = 1D Stripe
X LEDs = 2nd Segment = 2D Matrix
Even better would be being able to vary and repeat this subdivision endlessly (of course in the range of possibilty of total LEDs / PIN)

**Describe alternatives you've considered**
As far as I've seen it's also not possible using different PINs, since the option is always neither 1D OR 2D.

Would also happy to be able to only select the segment with the 2D Matrix for LEDfx support.

blazoncek, 2024-03-11:
You can set up matrix followed by strip. If you have physically reversed set up (strip followed by matrix) use reversing at the bus level and then (if necessary) set segments with reversing accordingly.
Please see [KB](https://kno.wled.ge/advanced/2d-1d-Mixed-Setup/).

closed 2024-03-11 by blazoncek

TheLastMuri, 2024-03-12:
omg, totally overseen it within WLED as well as in the documentation. 
Sry but still thanks a bunch!