wled/WLED #3813: Fix the Drift animation

BaptisteHudyma, 2024-03-12:
### What happened?

Hello !
The drift animation does not work as expected, only the lower right quarter is displayed.

### To Reproduce Bug

To reproduce: try it.

### Expected Behavior

The whole animation is displayed, instead of just a quarter of it

### Install Method

Self-Compiled

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

Other

### Anything else?

The bug is due to a rounding error with a cast to uint16_t instead of int16_t, MR to come

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BaptisteHudyma, 2024-03-12:
I opened MR #3814 to fix this issue

blazoncek, 2024-03-22:
Implemented via #3816

closed 2024-03-22 by blazoncek